betaflight/betaflight #1198: Use Huffman compression for blackbox data transfer

martinbudden, 2016-09-20:
This is a follow on from issue https://github.com/betaflight/betaflight/issues/1156 - Suggestion: Use compression to speed up on-board flash blackbox transfer

martinbudden, 2016-09-20:
See PR https://github.com/cleanflight/cleanflight/pull/2417 for an implementation of Huffman encoding and decoding. Decoding is in the test code and will need to be ported to javascript to run in the configurator.

@etracer65 and @mikeller - comments are welcome.

mikeller, 2016-09-20:
Thanks, @martinbudden. I have reviewed it, looks good to me. My only concern is that it will increase the RAM requirements, so we will have to decrease the block size for F1 targets. (We might be able to work around this with an implementation that supports concatenation of blocks of compressed data.)

martinbudden, 2016-09-20:
Yes, it increases RAM requirements, also code and data require about 1K of ROM. The simplest solution is probably not to include it in F1 targets.

@mikeller , will you be able to convert the decompression code into javascript and include it in the configurator?

mikeller, 2016-09-20:
> Yes, it increases RAM requirements, also code and data require about 1K of ROM. The simplest solution is probably not to include it in F1 targets.

Sounds reasonable. How much more RAM do the > F1 targets have?

> @mikeller , will you be able to convert the decompression code into javascript and include it in the configurator?

I could. But I have kind of given up on doing work with the configurator. @nathantsoi seems to have his own ideas on where he wants to go with the configurator, and @borisbstyle seems to be willing to take it this way, so I'll leave my fingers off it.

deepimpactio, 2016-09-20:
Don't know of many F1 targets with onboard memory.

etracer65, 2016-09-20:
@deepimpactio CC3D and Naze32 rev6 both have 2MB on-board flash.

AndersHoglund, 2016-09-21:
My Naze32rev5 has 16MB of on-board data flash. But I can live without compression.

brycedjohnson, 2016-09-21:
Could the huffman compression data be stored on-the-fly in the blackbox itself instead of just for the transfer?  More data stored instead of quicker transfer?

digitalentity, 2016-09-21:
@martinbudden what level of compression do you reach with 8-bit table?

martinbudden, 2016-09-21:
@brycedjohnson , yes, compressing data as it is stored is also a possible option. It might even allow blackbox data to be stored at higher rates, since it is likely that the flash chip is the bottleneck, not the processor - at least on the faster FCs.

I'm doing compression on the transfer first, however, as it is the easier option to code.

@digitalentity , I'm getting about 30% compression on the black box logs I've tried. They may not be representative though. Can you send me a couple of your logs, so I have a wider source to analyse?

medvdv, 2016-10-03:
Sorry to interupt but seems that huffman is not the best compression for blackbox data. Kind of adaptive delta compression or something like will be better considering nature of data, probably.

mikeller, 2016-10-09:
@medvdv: I think you are right for the case when data is compressed at the time it is generated, as at that time it would be easy to use adaptive delta on a per source basis.

This pull request implements compression at the time that data is read from storage on the FC, and sent to the configurator. Decoding the data into individual frames at this point, in order to use adaptive delta on individual streams would be very hard, and potentially slower than just use less effective, but not frame aware, Huffman encoding.

medvdv, 2016-10-11:
Oh, I see. It's just for transfer better trough usart. So you right, thanks for explanation i thought it was about storing more onboard

mikeller, 2016-10-11:
@medvdv That would be another idea, and actually even more beneficial. The current idea is to compress it before transfer, and decompress it in the configurator before storing it into a file. If it was compressed on generation, the decompression would have to be done in the log file viewer, since the configurator is agnostic to the internal file format for blackbox log files.

mkschreder, 2016-12-09:
For your information huffman has huge memory requirements compared to a handful of other techniques available for memory constrained devices. See p12 of this report: http://www.win.tue.nl/~mstolikj/publications/CS-12-10.pdf

I'm actually looking into this myself. Going to be putting together something that makes sense soon. I think best result can be achieved by examining recurring patterns in blackbox delta state and and picking an algorithm that deals best with that particular kind of data.

mikeller, 2016-12-09:
@mkschreder Feel free to submit a pull request. Here in Betaflight we are always happy to accept community contributions, and try to help out with them as much as we can. For more immediate communications, go to http://www.betaflight.tk/

martinbudden, 2016-12-09:
@mkschreder  , thanks for commenting.

The table you refer to is the memory requirements for decompression, not compression.

Compression is done on the flight controller, decompression is done in the configurator: it is the memory requirements of compression that are of importance.

The Huffman compression proposed for betaflight has much smaller memory requirements. 344 bytes for the code and 768 bytes for the Huffman table.

We have tested a number of different blackbox files and the relative character frequencies are fairly constant, so it is possible to generate a single Huffman table to be used for all black box files and still achieve approximately 30% compression.

I doubt you will find a more efficient option than proposed, but I am of course interested if you do so.

mkschreder, 2016-12-09:
Ok that sounds good.

I have a working simulator now that also supports simulated msp (basically
I can fly in the sim and connect to the simulated drone using the standard
configurator to configure it). I have an idea that it would also be cool to
replay blackbox data directly in the sim for a better feel as to what
happened during a recorded flight. Keep me updated on your progress.

mikeller, 2016-12-10:
@mkschreder The simulator sounds interesting. Would you mind sharing it with us?

mkschreder, 2016-12-12:
@mikeller: I pushed the code and instructions for building it to this repo: https://github.com/mkschreder/ninjasitl.git

Don't expect much from it. It is just a simple simulator that loads quake maps, does collision detection and simulates physics using bullet physics engine. It is by no means a complete product yet. 

The simulated msp is part of ninjaflight. If you build the SITL version of ninjaflight you will be able to access msp port through a pseudoterminal on linux instead of /dev/ttyUSB*. MSP port will be accessible at /dev/pts/* while simulated flight controller is running and can be accessed using the configurator just like an ordinary serial port. Simulator prints out it's pts name when it starts up and loads the flight controller. You will need to use that name in the configurator so you don't accidentally try to connect to a pts that belongs to one of your console terminals. 

You will also need a pak0.pk3 from quake3 to render the default map (pak0 is about 500 Mb so it is not included in the repo). If you follow instructions in the readme, when you start the simulator for the first time it will try to download the pak0 from a server online.

mkschreder, 2016-12-12:
A few things of particular interest is the sitl interface documented here: https://mkschreder.github.io/ninjasitl/structfc__sitl__client__interface.html

And also the system call interface between the flight controller and the rest of the system (between completely portable part of the code and the hardware/drivers): https://mkschreder.github.io/ninjaflight/group__syscalls.html . So things like micros() become system calls to a specific instance of the system and called like this: sys_micros(system). In the case of multiple flight controller simulated at the same time this allows you to make the system call to a specific instance of a system (ie when you do a pwm system call to write motors, this type of design allows the value to end up in the correct quad instance in the simulator). 

The hardware specific part of the code then just needs to implement all the system calls and the flight controller can be ported to any system.

mkschreder, 2016-12-30:
@martinbudden how about this scheme for packing blackbox: 

```
/**
 * Recursively pack changes between two equal size blocks into a delta buffer.
 */
static int16_t _delta_encode_block(const void *_prev_state, const void *_cur_state, size_t data_size, void *_output, size_t out_size, size_t block_size){
	// get char pointers to the data
	const uint8_t *prev = (const uint8_t*)_prev_state;
	const uint8_t *cur = (const uint8_t*)_cur_state;
	uint8_t *output = (uint8_t*)_output;
	// store header so that we can update it after we have encoded a block
	uint8_t *header = output++;
	// data size may be zero for whatever reason. In that case we do not write any delta data
	if(!data_size) return 0;
	// it is necessary to make sure we reset the delta bits
	*header = 0;
	// last iteration will have block size one
	if(block_size > 1){
		uint8_t bit = 0;
		//printf("encode %lu bytes with blocksize %lu\n", data_size, block_size);
		for(size_t c = 0; c < data_size; c += block_size){
			size_t len = data_size;
			if(block_size < data_size) len = block_size;
			// last block can be less than a full block size so we need to account for that
			if((c + block_size) > data_size) len = data_size - c;
			//printf("do block recursively at %lu of len %lu at output %p\n", c, len, output);
			// descend down into successively smaller blocks sizes while keeping track of which blocks have changes in them
			int16_t r = _delta_encode_block(
				prev + c,
				cur + c,
				len,
				output,
				out_size - 1, // -1 is for the header
				block_size / 8 // blocks size is always subdivided such that we can full in 8 bits
			);
			// if the encoding has failed we just return -1
			if(r < 0) return -1;
			else if(r > 0) {
				*header |= (1 << bit);
				out_size -= r;
				output += r;
			}
			bit++;
		}
	} else {
		//printf("encode %lu bytes at %p\n", data_size, cur);
		// this is the deepest iteration where we simply encode 8 bytes (or less)
		for(size_t c = 0; c < data_size; c++){
			if(cur[c] != prev[c]){
				*output++ = cur[c];
				*header |= (1 << c);
			}
		}
	}
	// if no blocks are dirty here then we don't write anything
	if(*header == 0) return 0;
	// return number of bytes written into the output
	return output - (uint8_t*)_output;
}

/**
 * Decodes a delta block and returns the number of characters consumed from the input buffer
 */
int16_t _delta_decode_block(const void *_delta, void *_output, size_t out_size, size_t block_size){
	const uint8_t *delta = (const uint8_t*)_delta;
	const uint8_t *header = delta++;
	if(block_size == 0) return 0;
	if(*header == 0) return 0;
	for(uint8_t bit = 0; bit < 8; bit++){
[... 36 lines cut ...]

Several deltas can easily be written after eachother into a stream and decoded in that order (actually this scheme is more efficient than my previous config delta/eeprom emulation implementation so I may look into using it for saving config changes as well). Each delta, when unpacked will update the bytes that have changed. 

Unit test (and example usage) here: https://github.com/mkschreder/ninjaflight/blob/master/src/test/unit/encoding_unittest.cc#L222-L372

Test compression figures for randomly filled buffer are as follows: 

> size: 472, fill (%): 0, compressed size (%): 0.00
> size: 472, fill (%): 10, compressed size (%): 22.03
> size: 472, fill (%): 20, compressed size (%): 34.53
> size: 472, fill (%): 30, compressed size (%): 44.92
> size: 472, fill (%): 40, compressed size (%): 56.78
> size: 472, fill (%): 50, compressed size (%): 65.89
> size: 472, fill (%): 60, compressed size (%): 78.81
> size: 472, fill (%): 70, compressed size (%): 84.75
> size: 472, fill (%): 80, compressed size (%): 93.43
> size: 472, fill (%): 90, compressed size (%): 102.54
> size: 472, fill (%): 100, compressed size (%): 113.98

Advantages: it's pretty fast and very simple. 
Possible disadvantages: there could be a better way. 

In the real world buffer delta is rarely random. Usually only some values change like gyro etc. These are usually grouped together in a single block so compression ratio is much better. It basically ends up with mostly just changed bytes and only a few control bytes. Either this is good as is or maybe it can be used as first step to pack a delta before further compression using some other algorithm. 

Comments are welcome.

martinbudden, 2016-12-31:
@mkschreder , your packing code seems like a reasonable approach. As you point out, real world data is not random, and blackbox data is not random at all. So I'd be interested in seeing what levels of compression you get with actual blackbox data. Your compression scheme may have a good fit, or it may not - testing with real data will show how good the fit is. I would say that if you are getting about 25% compression or better then your scheme is OK to adopt.

mkschreder, 2016-12-31:
@martinbudden: my blackbox rate with real data is about 2-3kbs compared to
same data using cleabflight packing scheme that resulted in 10kbs (side by
side).. The only special condition in my tests is that I currently log only
a limited set of fields. Compression does depend on sparcity and amount of
changes in the data. The only way to really know is if it is tested
independently on different setups and compared that way. I still think it
will do a lot better than current scheme in every sense because fewer bytes
are wasted for controln bytes.

martinbudden, 2017-08-01:
Implemented in PR https://github.com/betaflight/betaflight/pull/3000

closed 2017-08-01 by martinbudden