betaflight/betaflight #1212: micros() returning out-of-order timestamps?

jflyper, 2016-09-23:
I'm experiencing a strange phenomenon, in which successive call to micros() from an exti handler seems to return out-of-order timestamps.

Here's a pseudo code:

```
uint32_t t[5];

void extiHandler (extiCallbackRec_t* cb)
{
    if (tcount < 5) {
        t[tcount++] = micros();
    }

    if (tcount < 5)
        return;

    // Disable interrupt when done
    EXTIEnable(myIO, false);
}

void startMeasurement(void)
{
    tcount = 0;
    EXTIEnable(myIO, true);
}
```

It basically takes 5 consecutive interrupts and record the time. Examining the t[] array after the sequence frequently reveals that a pair of the timestamp are in the reverse chronological order.

Any thoughts?

ledvinap, 2016-09-23:
Micros() may return invalid time when called with basepri higher than systick

jflyper, 2016-09-23:
Yeah, that's what I was afraid of by looking at how micros() is observing SysTick carry over.

It may explain my case, as well as occasional glitch from the sonar measurements.

ledvinap, 2016-09-23:
It is possible to fix it, but is't not that easy. One possibility is to disable systick interrupt (increase BASEPRI enough) and handle overflow micros (something like https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/timer.c#L151)

Better varian it to implement two micros variants, one for interrupt enabled (~ current version), other for interrupt disabled (check overflow flag and handle it) and call correct version in `micros()` (and use correct version when you know context and it is time critical)

jflyper, 2016-09-24:
Hmm...

I can't figure out the SysTick's priority. In lib somewhere?

ledvinap, 2016-09-24:
https://github.com/betaflight/betaflight/blob/master/lib/main/CMSIS/CM1/CoreSupport/core_cm4.h#L1682
lowest available ..

androck, 2016-09-24:
Maybe it is also useful to disable your EXTI before each call to micros?

ledvinap, 2016-09-24:
@androck : BASEPRI is set to ISR level, so ISR can't interrupt itself ..

jflyper, 2016-09-24:
Since the SysTick runs at the lowest priority, it is (almost?) always unsafe to call micros() from ISR.

If that is the case, we should officially introduce a variant of micros() that is safe to be called from ISRs.

Thoughts?

borisbstyle, 2016-09-24:
Where exactly are the micros called from ISR?

I assume it is something you are working on?

digitalentity, 2016-09-24:
Sonar driver calls `micros()` from EXTI handler

jflyper, 2016-09-24:
Yes, it's in my code, and there is an occurence in the sonar_hcsr04.c, too.

```
void hcsr04_extiHandler(extiCallbackRec_t* cb)
{
    static uint32_t timing_start;
    uint32_t timing_stop;
    UNUSED(cb);

    if (IORead(echoIO) != 0) {
        timing_start = micros();
```

There may be others.

ledvinap, 2016-09-24:
IMO it's better to implement both together with wrapper (`micros()`) that decide which one to call. Using two functions in code will introduce bugs eventually ...

jflyper, 2016-09-24:
Isn't it a burden to export SysTick priority out of core_cm4.h? Shall we do it by peeking SysTick's nvic settings?

ledvinap, 2016-09-24:
Setting priority after configuring frequency should do ... Or just assume lowest priority for systick, it shouldn't interrupt anything anyway ...

jflyper, 2016-10-07:
The active PR is #1264.

I'm closing this now.

closed 2016-10-07 by jflyper