betaflight/betaflight #1215: Sbus not working.

dannygravity, 2016-09-23:
HI I'm having issues with setting up  my kiss with bf 3.0 using sbus, it just don't work if I reflash the  kiss back to rc28   with out changing anything it will work on sbus but soon as I flash bf it don't work   I than have to change it to ppm than it works.
any idea guys/ brois

lichtl, 2016-09-24:
Betaflight on KISS FC require a different pin for Sbus. You need to use the RX one on the side

dannygravity, 2016-09-24:
Can you tell me with one please

borisbstyle, 2016-09-24:
I believe it should work on the same pin. Just needs correct uart
configuration

lichtl, 2016-09-24:
I haven't tested BF on the Kiss yet but based on the port from Nathan it seems that the this is not an UART pin.  http://nathan.vertile.com/blog/2016/07/29/betaflight-kiss-flight-controller/#serial-rx

My understanding this are the UART pins.
UART1 - RX/TX next to micro USB
UART2 - should be the one at Aux1 (used for Telemetry on Kiss)
UART2 - TLM pad for ESC's

dannygravity, 2016-09-24:
I have tried all the uart but I will retry again fingers crossed

borisbstyle, 2016-09-24:
It certainly worked for me a month back when I was testing it with my X4RSB

dannygravity, 2016-09-24:
Using the std pad (wiring) as kiss or do I need to use the telemetry pad if
I load up kiss fw it works??

dannygravity, 2016-09-25:
Just to confirm I have now tried the different way we talked about and it
still not working.

basdelfos, 2016-09-30:
It works, have it working on both KISS FC's with BF 3.0. Only the default configuration after flashing BF 3.0 is wrong. Serial RX should be configured on UART2 instead of UART1 (i think is the default). You can use the normal SBUS pads like with the KISS firmware.

PR #1285 (2016-10-13): Moved KISS default serial RX to USART3 on @basdelfos' request.
Fixes #1215.

@basdelfos can you please confirm that this defaults to the right UART now?
closed 2016-10-13 by borisbstyle