betaflight/betaflight #1219: PikoBLX: FrsKY/SmartPort Telemetry VFAS not working

RipperDrone, 2016-09-24:
Have a new Piko, running FrSky telemetry over FrSky XSR receiver. VFAS voltage always shows 33.0 - telemetr_inversion = ON as per PikoBLX manual. Other values like RSSI come in fine, so telemetry serial is obviously working.

Manual says voltage is scaled already so I left scaling value at 100. When I lower it, PikoBLX start beeping alarm, probably due to low voltage detect?

Ideas?

brycedjohnson, 2016-09-29:
RSSI doesn't get transferred from the FC, it is directly from the XSR to the taranis.  Voltage works well for me using some of the small D8 frsky receivers over frsky (not smartport) telemetry.

RipperDrone, 2016-09-29:
I see. How can I best check then whether or not the telemetry transmission from FC to Taranis is working ok?

RipperDrone, 2016-09-29:
Thank you, and yes: I do get asterisks and many sensors updating, but VFAS always shows 36.3V. Must be a hardware issue then I guess.

RipperDrone, 2016-09-30:
@brycedjohnson You do get the correct voltage shown in BetaFlight Configurator when USB-connecting the FC to your computer, don't you? Which voltage scale setting are you using, something around 100?

If yes, there must be a defective hardware issue with my PIKO and we can close this issue here...

closed 2016-10-01 by blckmn