betaflight/betaflight #1224: Add Altitude to OSD tab

Jeffbabb5, 2016-09-25:
Please add an altitude switch to the OSD tab so altitude can be shown with boards that have a barometer.

Thanks.

nathantsoi, 2016-09-25:
This is already done. Coming in the next release. Please try the latest
build from master until then.

jflyper, 2016-09-27:
Kinda noob question.

Do we have to use an updated configurator, too?
BFC 1.8.1 does not draw OSD tab correctly after some post-3.0.0 PR.

mikeller, 2016-09-27:
@jflyper: What board are you using? Currently only a few boards support the on-board OSD that is catered for by the OSD tab, for the rest of the boards, the configurator throws exceptions if the OSD feature is activated and the OSD tab selected.

jflyper, 2016-09-27:
Omnis! :)

The problem was solved by fetching BFC and the chrome developer magic.

I believe we should recommend BFC upgrade together when we point to non-release snaps, especially the OSD related stuff.

nathantsoi, 2016-10-02:
yeah, new configurator (master @ https://github.com/betaflight/betaflight-configurator) should be working with the latest on master in https://github.com/betaflight/betaflight

martinbudden, 2016-12-20:
Closing since feature added.

closed 2016-12-20 by martinbudden