betaflight/betaflight #1226: SPI with multiple slaves on same channel

sylarswe, 2016-09-25:
Hi!

I have been trying to add a second slave to the SPI2(flash) channel but with no success.
Both devices works separately on the bus when disabled in the code.
They are connected with SCK,MOSI connected between both devices(flash and RTC6705) and SS on separate pins. Is this possible with current code or is this something that has to be added to the drivers ? Or have I just missed something ?
It is an F3 target

#define USE_SPI_DEVICE_2 // PB12,13,14,15 on AF5
#define USE_SPI_DEVICE_3 // PA15,PB3,4,5

#define USE_FLASHFS
#define USE_FLASH_M25P16

#define M25P16_CS_PIN           PB12
#define M25P16_SPI_INSTANCE     SPI2

#define MPU6000_CS_PIN          PA15
#define MPU6000_SPI_INSTANCE    SPI3

#define VTX
#define RTC6705_CS_GPIO         GPIOA
#define RTC6705_CS_PIN          GPIO_Pin_14
#define RTC6705_SPI_INSTANCE    SPI2

Thanks in advance

jflyper, 2016-09-26:
RTC6705 is rumored to be ill behaving in the sense that it always listens and responds to the MOSI (SPIDATA).

My theory is that the SS (SPILE) seems to be only used for delimiting each register transfer.

You may verify this theory by connecting only the RTC6705 and drive it without asserting/deasserting the SPILE.

My guess is that if you want the RTC6705 as a slave on a shared SPI bus, you have to use a tristate buffer on the SPIDATA line with enable connected the SPILE line.

sylarswe, 2016-09-27:
You´e totally right! the RTC6705 even listens sometimes with just MOSI (no SPCLK or SPILE/SS), Thanks for that info!

But I have noticed one more thing, I cannot define RTC6705_CS_PIN(SPILE) to whichever pin I want, it only works on dedicated SPI_NSS pins from the datasheet which is weird when I can define both M25P16_CS and MPU6000_CS_PIN to whichever GPIO pin I want without any issue at all.

However when I defined MPU6000_CS_PIN PA14 and RTC6705_CS_PIN PA15 everything works, then I decided to add the flash to the SPI2 bus also with M25P16_CS_PIN PB12 and everything stops working BUT when I hooked ut up to my oscilloscope I found something interesting....
firstly SS is pulled low on PA15 to set channel to VTX, mosi and sck does its works and then PA15 goes high again....... and then when FC shall initialize flash both PB12 AND PA15 are pulled low so there was not even a chance that the RTC6705 would even listen :)

Solved this issue by replacing "(SPI_I2S_GetFlagStatus(instance, SPI_I2S_FLAG_TXE) == RESET)" with "(spiIsBusBusy(instance))" in the bus_spi.c driver, after that it does not involve PB12 PA15 so thats a start.....

borisbstyle, 2016-12-20:
Closing this one, but discussion can continue if necessary

closed 2016-12-20 by borisbstyle