betaflight/betaflight #1229: beeper

janlebelg, 2016-09-26:
on naze32 rev 5 the beeper only beeps once in stead of repettly 
when affecting in modes a beeper button
it is hard to retrieve a model in this condition

mikeller, 2016-09-26:
What version of the firmware are you using? There was a bug in the beeper code in 3.1.0 until very recently.

janlebelg, 2016-09-26:
Hi

It is the offisial naze 3.0.0 11-09-2016 load in betaflight
Regards
Jan

blckmn, 2016-11-19:
Can you please test: https://github.com/betaflight/betaflight/pull/1599

janlebelg, 2016-11-19:
i love to test 

bud i am a noob in compiling s

o is it posible to send me the "betaflight_3.1.?_NAZE.hex" file ore link please

i wil test in 24H aftre reseaving the file

regards jan

mikeller, 2016-11-19:
http://andwho.sytes.net:8080/job/BorisB_BetaFlight/730/artifact/obj/betaflight_3.1.0-730-d9a8655.zip

janlebelg, 2016-11-21:
beeper ok 
did no maiden flight
it is reaning kats and dogs her
bud I dit not see anything wrong 
gread job

tested this vertion
# Betaflight/NAZE 3.1.0 Nov 19 2016 / 22:23:07 (d9a8655)

closed 2016-11-21 by janlebelg