betaflight/betaflight #1238: Clear onboard flash (erase blackbox)

mixMugz, 2016-09-28:
May be possible make stick command (or aux command) for clearing onboard flash memory for writing blackbox log from start?

Now i need each time to connect to computer for format flash.

Gamma-Software, 2016-09-28:
Try to post a request instead, what you're asking for is not an issue.

Austinthemighty, 2016-09-30:
i will try to add it.

UltiFix, 2016-10-14:
This is under the future improvement sectin

fftunes, 2016-12-31:
Where is the "future improvement section" ?

Would like this too. Stick command, or maybe an extra "flight mode" for erasing flash? TX could be programmed to default to this at start. With a beep at the end to confirm when done.

It could allow to keep blackbox running all the time.

commit a373c095 (2015-08-20): Correct new rxfail documentation.  Fixes #1238
closed 2018-04-21 by DieHertz