betaflight/betaflight #1241: Says It works with Rc Explorer F3 But still dosent have the special tail code.

jrach, 2016-09-30:
If anyone knows how to implement the triflight code into betaflight that would be awesome. I don't understand a thing about coding. Thanks! Happy flying!
- Jacob Rachwal

DerOrakel, 2016-09-30:
The RC Explorer F3 is just another F3 Flight Controller, you can use it for pretty much anything. If you wan't to use it in a tricopter go with Triflight. If you wan't to use it in a Quad like I do use BF or anything else that supports it.

mikeller, 2016-10-11:
@jrach: Also, version 3.0 of Betaflight works quite well for tricopters, with minimal tuning. Do you have anything in particular that gives you trouble when using Betaflight on a tricopter?

martinbudden, 2016-10-12:
@mikeller , although I haven't looked at the Triflight code in any detail, I believe it has some useful enhancements that could be merged into betaflight. We should work with @lkaino to get these merged into betaflight.

mikeller, 2016-10-12:
@martinbudden: It has, I found the ability to calibrate the tail servo center / min / max by using sticks on the TX quite useful. When it comes to flight controller performance, their code seems to be optimised to run on the RCExplorer models (they even say so in their howto videos), and I found this to be true. I don't know how much value we'd get from integrating this into Betaflight?

mikeller, 2016-10-12:
@martinbudden: Totally agreed. The tuning part alone makes setting up a tricopter a much more pleasant experience, compared to fiddling with the inputs in the 'Servo' tab in configurator. I think in terms of PID controller, their main difference is that they have a tail mixer that is implemented specifically as a driver for a tail servo, and separate from the 'normal' PID controller. They also support servo position feedback directly into the FC.

My experience from alternatively trying Betaflight and triflight on a tricopter over a period of time was that Betaflight used to be very fiddly and difficult to get the yaw PIDs right, whereas triflight was more stable on the tricopter, except for things like servo reversal not being possible if your tail servo wasn't oriented the same way as the RCExplorer one was. Once Boris implemented the new PID controller, Betaflight went to being more stable than triflight on the tricopter, and yaw PID tuning became much less sensitive to oscillations and much less fiddly.

lkaino, 2017-06-22:
I have in plans to start the merge discussions later this year. Triflight is already based on Betaflight so once the time comes creating the pull request(s) should be rather simple.

The Triflight new feature development is still ongoing, I want to stabilize the performance before the merge. There is an active prototyping and testing community at RCExplorer forums.

There's also missing documentation to write and finalize the unit tests.

closed 2018-04-21 by DieHertz

SJ-Innovation, 2020-07-24:
The Triflight project seems dead. And Ikainos merge was never completed nor merged. I have taken on the challenge of porting Triflight over to Betaflight 4.3 (as is currently on master)