betaflight/betaflight #1245: Motor output issue on latest update (3.0.0)

thedreygen, 2016-10-01:
Hi, I am new to the Naze controller, so hopefully it is an easy fix that I have happened to overlook. 

I recently bought a naze32 to upgrade from my old cc3d. I had it working fairly well and was doing some tuning today. I then realized there was a new major firmware out so I flashed it onto my FC. After flashing and recalibrating the accelerometer, the motors won't spin in response to my RC inputs or to the testing sliders on the "motors" tab. The motors spin up only if I have battery power connected and reset the board via usb, however, I have no control over the motor speed and it seems to be constant.

I am using a NAZE32 REV6a MPU6500 32-bit 6 DOF, and flashed the NAZE 3.0.0 firmware. I guess I will go back to an older firmware and see what will work for me. 

Thanks!

thedreygen, 2016-10-01:
*update*
The 2.9.1 release works just fine with RC or computer control of the motors/throttle. Not sure what would or could cause the error, but I am glad that it seems to be something in the firmware (and not something that I broke).

Cheers, and thanks for all the time put into this project!!

borisbstyle, 2016-12-19:
Please check your CPU usage. I expect it was too high preventing it to arm

closed 2016-12-19 by borisbstyle