betaflight/betaflight #1250: Camera angle compensation reduces angular velocity.

GrishaLutsenko, 2016-10-01:
When enabled ( I had it at 25 degrees) and max corner velocity (around all axis) 900 degree - with no angle compensation I had that speed, but when enabled rotation speed was reduced to about 300 degrees ( subjective, maybe a little different number but definitely much less than with disabled camera angle compensation)

borisbstyle, 2016-10-01:
Yes this is normal as yaw cant deliver those speeds. when you mix yaw and roll the yaw becomes a limit.

closed 2016-10-01 by borisbstyle

GrishaLutsenko, 2016-10-01:
Yes, but yaw shows same 900 degree in PID settings page and when compensation is disabled yaw makes about declared values. That's why that looked strange for me

borisbstyle, 2016-10-01:
@GrishaLutsenko Its the physical limitation in quads not the software limitation ;)

GrishaLutsenko, 2016-10-02:
Ok... Just still not clear for me, why if yaw can do that separately at 900 degrees and roll can do, why can't do that together :(

borisbstyle, 2016-10-02:
Yaw cant do that seperately. Also besides speed the yaw has much slower
acceleration and deacceleration than roll and pitch.

GrishaLutsenko, 2016-10-04:
Btw today found one more problem around camera angle compensation.
As I mentioned before turning speed is reduced, but... At some points turning yaw about 70% of stick value and small roll change it makes sudden full speed rotation and result is not predictable...