betaflight/betaflight #1271: LED Strip indicator mode not work correctly.

mixMugz, 2016-10-05:
Try to make led strip with 16 leds.

0 1 2 3
4 5 6 7
8 9 10 11
12 13 14 15

Set led 0 3 12 15 to indicator mode.

When you roll or pitch - leds not blink correctly. 
1. Blink only when you roll on right or pitch down
2. Not blink other location.

mikeller, 2016-10-10:
Can you add a screenshot of your LEDSTRIP config tab? Configuration for the indicator is a bit involved, and it's easiest to help you if I can see what you've got already.

mixMugz, 2016-10-22:
[image]

mixMugz, 2016-10-22:
Also check larson scanner on this config. Its table led grid 4x4

gaelj, 2016-10-22:
Indicator function requires orientation to be set on each LED.

mixMugz, 2016-10-23:
> Indicator function requires orientation to be set on each LED.

Can you tell me please how to set up my configuration for property blinking?

mixMugz, 2016-10-23:
Just checked now. Set up different led orientation for each led with indicator overlay and nothing. All leds blinking only on pitch down and roll right. No reaction on pitch up and roll left. Please fix it

mikeller, 2016-10-23:
So you set the orientation (under 'LED orientation and colour'), so that the correct little direction arrows are shown for every indicator LED?

mixMugz, 2016-10-23:
There is my last config. Still nothing. Work as previous message. (leds blinking only on pitch down and roll right)
[image]

mixMugz, 2016-10-23:
Also. Why larson scanner for led 5,6,9,10 work only from 5 to 6 to 9 to 10, instead from 9 and 10 to 6 and 5?

mixMugz, 2016-11-22:
So. Any news?

mikeller, 2016-11-22:
Not really. Looks all correct to me. I was hoping that @gaelj could shed some light onto this (no pun intended).

gaelj, 2016-11-22:
Hi, simply because it was implemented by making the animation follow the
LED numbering.

mikeller, 2016-11-22:
@gaelj but do you have any clue as to why @mixMugz ' indicators are not working as they should? For what I can tell from the screenshot, the orientation is set correctly.

closed 2016-12-19 by borisbstyle

slamblam, 2016-12-24:
I am having the same identical issue, west side is not working...

mixMugz, 2016-12-25:
Closed? Why?!