betaflight/betaflight #1274: Revo F4 Unknown "Twitching" Issue BF 3.0.0

Chulz, 2016-10-05:
Having some "twitches" come out during my attempt at tuning today however I do not feel they are the fault of the PIDs. Heres the specs on the quad and then I will attempt to concisely state issues I have looked into already:

Craft: RuXus 5", RotorGeek 2206 2350kv, Aikon SEFM 30A, RaceKraft 5045x3, XSR SBUS, OP REVO F4 running BetaFlight 3.0.0 with Gyro Refresh 8kHz, PID Loop 8kHz, and MultiShot with BLHeli_S which has been calibrated.

I have created both a video and a BlackBox log for which I will insert at the end but from what I'm seeing in person and in the log there are decent spikes of random value and random motor around 1340-1370 throttle input that have nothing to do with PIDs as they stop above roughly 1400. I have attempted adjusting PIDs, running stock config settings on BF, reflashing BLHeli_S, recalibrating ESCs, running different Gyro/Looptime/ESC Refresh rates (all while never making the board faster than the ESC) to no avail. All solder points are solid, no screws are touching the stators of the motors, etc. Build wise I believe the setup is sound, no frame conduction at any point. The vibrations/twitches seem constant at that throttle input described. Please view the video and the corresponding blackbox log. In the video the twitches are not only visually identifiable but also audibly. Any help is greatly appreciated, I'm pretty stuck at this point.

Best Regards,
Brent
Video: https://vimeo.com/185700292
[image]
[image]
BlackBox Log File: 
[BL Log 2khz.TXT](https://github.com/betaflight/betaflight/files/512182/BL.Log.2khz.TXT)

rodgerramjet, 2016-10-06:
Does it happen on unsynced mode?  I had a similar problem with multishot on synced- same escs but f303 fc. 

Didnt happen on bf 2.9, didnt happen with oneshot125 on bf 3.0- only when i used synced multishot

Chulz, 2016-10-06:
Prior to this upgrade I was running a Naze Rev5 which was severely pushed to its max when running Betaflight 3.0.0 so I wanted a chip that could handle it plus lower looptimes and the like, thus the Revo F4.

I tried running OneShot125 and 4kHz together on this Revo on 3.0.0 and I got the same result, I just didn't want to make the post extremely long with all of my BlackBox logs. Issue points to Revo or Aikons or maybe the use of both. I have yet to try RaceFlight but I very much like BetaFlight and hope to stay with it.

What was the fix listed if you don't mind me asking? Other than attempting to reverting back to 2.9.1 that is.

andre92, 2016-10-06:
you have a super noisy yaw gyro, check it out in blackbox viewer with smoothing disabled (S key disables smoothing). set yaw(hz) to 80 and set up the gyro notch filter (set debug_mode=notch in cli to get raw gyro so you can get the Hz and cutoff, see wiki for more info)

Chulz, 2016-10-06:
@andre92 Thank you for another great response, I can now see that in the majority of my blackbox logs my yaw gyro is out of control. I am, however, seeing quite a variability in my pitch gyro as well across the logs but the yaw gyro (with smoothing disabled - pictured below) seems to ramp up first. Roll seems to be rather well contained. Would there be another aspect that requires change affecting the pitch gyro? I will educate myself on this as much as possible via the wiki and give it a try today and report back! Thanks again

[image]

andre92, 2016-10-06:
Yaw gyro twitches have the worst effect on your motors since yaw pids are highest. This makes the motors bounce between full and idle quickly and the pitch gyro may pick up on that. Yaw LPF may fix the motor twitching to some degree (try 80 hz), and gyro notch filtering may take care of remaining noise on the pitch gyro. Or try setting gyro_lowpass_level = HIGH for better gyro filtering.
Another possibility is to check for loose flight controller mountings, or maybe soft mount it to take care of the worst vibrations.

Chulz, 2016-10-06:
Multitude of fixes implemented much thanks to @andre92 and are as follows: I introduced soft mount rubber o-rings below the Revo FC on the standoffs (only bottom side for now). In addition I found that the median motor noise between both Pitch and Roll was between 120-185hz so I ultimately set a notch filter value around 150hz even though noise surrounding pitch was far worse, shown below:
[image]
In addition to this based on advice from @andre92 I also increased the Yaw LPF to 80hz. My results are shown below. I have taken pictures of the same throttle response and throw as prior and oscillations have all but disappeared.
Before:
[image]
After:
[image]
I definitely have PID tuning left to do and will save that for the field but as of right now I sincerely appreciate all of the help. Regardless I feel more educated on gyro filters and their overall use within the constructs of BlackBox. I will continue to read into filtering and its overall effects as I purchased this Revo F4 board to create the most locked-in quad possible as a challenge of sorts. Also thanks to @rodgerramjet for your contribution!

andre92, 2016-10-06:
@Chulz I'm happy to see your results. Could you post your BBlog from after tuning? Just for reference. Credit should go to @joshuabardwell, I learned everything about filtering from his youtube channel. I was just watching his latest livestream and noticed one more thing: Betaflight 3.0 apparently introduced some gyro twitches that weren't there in 2.9.1. This may be caused by the default gyro_lowpass_level setting, it is set to NORMAL while 2.9.1 used the equivalent of HIGH. That may cause issues if the notch filter is left disabled. 
I guess we are supposed to use NORMAL only when combined with notch filter, or HIGH if we do not use notch filter. Maybe @borisbstyle can comment better on this.

Chulz, 2016-10-14:
Here's the log after tuning out the gyro issues. The PIDs still need some work but unfortunately I'm encountering a new problem: The CC3D Revo resets every time there is a moderately significant impact. Whether it be a crash or simply a harder than normal landing it resets and thus the ESC reset as well. Not sure, haven't experience this with any other board, maybe the Revo is very prone to impact/vibrations?

Here's the log of the fixed problem:
[BL Log 1khz YLPF80 Notch 150.TXT](https://github.com/betaflight/betaflight/files/530265/BL.Log.1khz.YLPF80.Notch.150.TXT)

mikeller, 2016-10-14:
@Chulz:The resetting sounds more like a problem with a power supply solder joint / connection. Check all of these by wriggling on the cables and see if that causes a reset. Could even be tired pins in an XT60.

andre92, 2016-10-14:
@Chulz you could do with notch cutoff at 140 and notch_hz at 210 for both gyro and dterm, that is where the most motor noise is on your copter, according to your latest log. 

As for the rebooting, I suspect either a loose wire or a short. (I actually had a signal wire come loose on an esc, which shorted 5V to Gnd in flight) Check all connections with a multimeter (and check for shorts)
Maybe log VBat (if possible on Revo) so brownouts can be detected.

closed 2016-12-01 by blckmn