betaflight/betaflight #1276: Micro Scisky 32 flashes fine, but will not connect to GUI?

Armand9x, 2016-10-06:
I've tried flashing different versions and all flash successfully. When I go to connect it, it say connecting, then hangs and the connect button says "connect again"

I've flashed it with the bootloader shorted, as its the only was to flash it. 

Here is a video showing the issue.

https://www.youtube.com/watch?v=aCAcpqGM9is

The software works, as I am still able to bind my transmitter.

I've tried installing and reinstalling the VCP drivers.

Thoughts?

AndersHoglund, 2016-10-06:
This FC does not have any VCP. It seems to be a Naze32 clone/derivative and it that case it should probably use the Silabs CP2103 drivers. In the video it seems to connect just fine on COM4, even flashning works, so there is probably no issues with Windows drivers etc. More an issue on the MSP layer in that case... But just to be sure, check that you see the "Silabs CP210x.." driver under "Ports & LPT" in Windows device manager. If not, re-install. 
I don't have this board, cant be of much more real help.

Armand9x, 2016-10-06:
I tried reinstalling that driver multiple times now..

Also tried two computers, and several cables...

Weird that it flashes fine but can't access the GUI

AndersHoglund, 2016-10-06:
Not weird at all. 
Flashing is done by the STM Bootloader over COM4 and the Silabs CP2103 USB interface. If that works, all is well with drivers etc. The BFC GUI uses the same COM-port and USB interface but talks to BF firmware using MSP protocol. Something must be broken here. Restarting BFC and Chrome as well might help.
Have you tried any of the previous 3.0.0 RCs? Or any of the 3.0.1 CI/alpha builds? I.e. have you has it working before?

Armand9x, 2016-10-06:
I've had it working before.

I had it working with 2.9.1 and 3.0. 

During using 3.0 it stopped arming suddenly, then I went to change settings and it would NOT save at all. Then it stopped letting me access the GUI.

Something obviously stopped working because it was fine before..

I tried restarting Chrome and BFC, and my computer. Tried two computers and no luck.

borisbstyle, 2016-12-20:
I will close this issue due to long inactivity. If the issue still persists on 3.1 than please open a new issue with full detaills.

closed 2016-12-20 by borisbstyle