betaflight/betaflight #1277: Receive smartport telemetry

discoltk, 2016-10-07:
Please correct me if I'm wrong, but as far as I can tell betaflight does not currently receive telemetry from smartport.    I see in the source there is a function to read from the smartport serial, but I don't see any conditions where it would process the incoming telemetry.

If this is a correct assessment?   If so, what are the barriers to adding handlers for incoming telemetry?   I think most people would enjoy getting RSSI data this way, and I have a smartport vario that I'd like to get data from to provide to my OSD.   

Thanks in advance for any support on this topic!

jflyper, 2016-10-09:
With S.PORT (Smart Port), I believe that receiver act as a master, and polls sensors on the telemetry bus. FC acts as sensors and send back sensor data. No incoming telemetry data for FCs.

RipperDrone, 2016-10-09:
SmartPort telemetry over FrSky runs fine on my SPRAINGF3. BF 3.0.0...

mikeller, 2016-10-09:
@discoltk: I can see what you are trying to do, but I am not sure if this is supported in general, since the receiver (that has RSSI data) is master on the FC - receiver serial connection. (i.e. you'd have to use a second serial connection for the FC to be master on).

raphaelcoeffic, 2016-10-12:
In fact, there could be S.PORT data to be received by the FC, as S.PORT in a bi-directional protocol. This had not really been used up to now, but I think there are some good use-cases. OpenTX 2.2 branch (next) supports sending/receiving arbitrary data from a LUA script. Data sent from LUA are relayed by the transmitter module to the receiver, which in turn spreads it through S.PORT on the receiver side. This means that we can make a LUA script communicate freely with the FC over S.PORT.

The first use-case that comes to my mind and which I'd like to implement is the possibility to tune PIDs or possibly other FC parameters from my TX. This could be done fairly easily by coding a S.PORT<->MSP bridge into the smart port telemetry code. If you guys are ok with it, I'd love to take care of it :-)

discoltk, 2016-10-12:
@raphaelcoeffic  this is great news!   

I'm glad to help with testing or more if there is a working code example to follow.

I'll also provide a bounty of 0.25 bitcoin to add the function to read telemetry from s.port so that it can be used for things like OSD.

raphaelcoeffic, 2016-10-12:
@discoltk: do you mean from sensors which are additionally connected to the S.PORT bus?

discoltk, 2016-10-12:
@raphaelcoeffic My setup currently is frsky xsr connected to frsky vario, which is connected to an F3 FC on a dedicated UART.   I've bridged the tx/rx pins, but as you know, no dice with FC getting vario (or rssi) data.   I can get it on the taranis, and even pass it back up an AUX channel as a ghetto rigged version.   

Way nicer if the FC can either snoop on existing polling the receiver is doing, or otherwise take control of the target/sensor.  Maybe there can be some kind of virtual sensor that you can populate from data such as s.port or other addons.

Then providing this to OSD is much cleaner.

raphaelcoeffic, 2016-10-12:
I'm quite sure the RX does not mirror the RSSI packets on his wire, but I need to check. The vario should not be a problem. However, I'm quite unsure how this is supposed to be integrated into BF...

discoltk, 2016-10-12:
@raphaelcoeffic Maybe it can use its own option flag where you override the FC's vario / baro / whatever with the data from s.port in the case that flag is enabled?  Maybe later a more advanced abstraction of telemetry sources could be added.

raphaelcoeffic, 2016-10-12:
Sounds ok to me.

raphaelcoeffic, 2016-10-21:
@discoltk after trying out, it seems I was right: the receiver does not send its own packets (RSSI, etc...) on the wire, but only to the transmitter, over the air. Other sensors attached to the S.PORT bus should be readable without any issues. However, I'll focus on the S.PORT/MSP first, as this is something many people would like to have.

discoltk, 2016-10-21:
@raphaelcoeffic   Well, any improvement to the utility of the smartport interface is much appreciated.   If there is anything I can do to help the vario support along, please let me know.

Will the lua script interface be lower latency than the standard aux channel hack?

closed 2016-12-01 by blckmn

barthalamu, 2017-03-10:
I know I'm late to the party, but this is a feature I'd love to have!
I don't need RSSI just any external FrSky sensors. I currently have a RPM/Temp sensor, 150a hall effect sensor, and cell voltage monitor connected to my X4r. It would be awesome to have this data displayed on OSD as well as being logged in Blackbox!

A possible alternative would be to be able to send these telemetry values back to the FC over channels like the RSSI hack, but it isn't pretty and wastes channels and bandwidth.

Yamaford, 2017-04-21:
@barthalamu I'm ok with using a channel to feed this back to the FC. Does betaflight know what to do with the RPM data? Can it be logged in blackbox and would I retain the PID/VTX adjustment feature?