betaflight/betaflight #1280: REVO - LED Strip - 0 remain diodes

oglocdogg, 2016-10-07:
Hi,
I bought REVO FC, started soldering and configuring at betaflight 3.0.0, what I saw, I have 0 LEDs remaining.

Is there any option to enable it?
[image]

nathantsoi, 2016-10-08:
i think the led strip feature is new in betaflight 3.0.1, which will be out soon, or you can download it from the jenkins server before then. the LED pin will be on motor output 5

RobbieHood, 2016-10-08:
Running 3.0.1 here and can confirm that i can correctly program LEDs. I can not controll roll or pitch anymore though hehe....Can not seem to find the issue but thats the risk of compiling a pre release ^^

RobbieHood, 2016-10-08:
https://youtu.be/Q-lmUkftHOM

blckmn, 2016-10-08:
I'll close this issue as it is resolved in respect of LEDs.

closed 2016-10-08 by blckmn

mikeller, 2016-10-09:
@RobbieHood Re roll / pitch issue, double check your rates to make sure they have not been set to 0 during the update.

RobbieHood, 2016-10-09:
@mikeller thanks for the tip and will try it in a bit.
I get reaction from controlls in the gcs and the led indicators act accordingly. The motors simply won't play when asked.

RobbieHood, 2016-10-09:
No that did not work unfortunately.

RobbieHood, 2016-10-09:
Did a fresh pull fron github.
After that flashed with full chip erase and all seems to perform as should now. Need to do a test flight.