betaflight/betaflight #1286: Unable to restore srate default behaviour in BF3.0

skullq, 2016-10-10:
To check different stick feeling, I have  changed srate to 0 and increase rc late above 2 with some expo. After checking stick feeling I tried to restore it to default my srate without rc expo but stick feeling remained no srate until image flash.
I think, I looks buggy. Please check this one?

mikeller, 2016-10-10:
Are you flashing between 2.9.1 and 3.0.0? If so it's normal that rate / srate setting is lost during flashing and restoring, since the name has changed.

skullq, 2016-10-10:
There was no previous BF version installed because FC(SPRacing F3) was new one.
Recreation steps as below.
1. rc rate to above 2, srate to 0 and some expo.
2. flying.
3. restore your previous srate based setting (my setting was rc rate 1.2, srate 0.72 and no rc expo and diffrent taste of srate yaw setting)

andre92, 2016-10-10:
SPRacing boards have Cleanflight preflashed and may use different configuration storage, it could help if you use Full Chip Erase when you flash it with Betaflight 3.0 for the first time.

skullq, 2016-10-10:
That's really reasonable. I will test it one more time after full flashing.

closed 2016-12-19 by borisbstyle