betaflight/betaflight #1287: Motor wind up?

RobbieHood, 2016-10-10:
Hey devs,

I have a small problem on BF 3.0.1 on REVO target.

Today I decided to tune a bit and setup the MOTOR_STOP feature.
I programmed a 2 position switch to arm and a mixed 3 position switch for flightmodes.
The flightmodes being:
1:Angle
2:Horizon
3:Default
4, 5 & 6: Default+AirMode

Its all working great except for one thing I noticed coming from spinning motors when armed, is that when I give a bit of throttle on the modes that have MOTOR_STOP active the motors will start winding up. I think I read somewhere this should not be happening anymore?

This is my first none OpenPilot based firmware and GCS so I am still finding my way. Maybe it is something I should have read better and if so then my apologies for that ^^

Keep up the great work!

RobbieHood, 2016-10-10:
https://youtu.be/PDJNfWNGGRA

borisbstyle, 2016-10-10:
Thats airmode what you experience. It gets activated above certain throttle
threshold. It should be all described in the wiki and is normal behaviour

RobbieHood, 2016-10-10:
No sorry maybe I was not clear. This happens on flightmodes 1 to 3 which do not have airmode active.

borisbstyle, 2016-10-10:
Even without airmode there is a zero throttle stabilisation

set pid_at_min_throttle = ON / OFF

RobbieHood, 2016-10-10:
Ok thanks Boris. Keep up the great work!

closed 2016-10-10 by RobbieHood