betaflight/betaflight #1292: SP Racing F3 Baro and Mag not available in 3.0

vicewize, 2016-10-11:
SP Racing F3 Baro and Mag not available in 3.0, if I go one firmware down everything works, accelerometer seams a bit off or funky, or it could be me. Thanks!

andre92, 2016-10-11:
If you want to use MAG and Baro, you may need to set the gyro at a lower rate. Maybe the databus cannot keep up with sampling all those sensors. Check tasks and make sure task_accel is executed at a stable rate. (should be near 1000)
On my SP Racing F3 EVO (spi), I can only reliably run 4k/2k with everything enabled.

vicewize, 2016-10-11:
I see that makes sense, got it working with your help. I actually got it working at 2k/1k right now seams stable. Task at a stable rate 1009. Thanks.

closed 2016-10-12 by martinbudden