betaflight/betaflight #1308: Naze 32 Xrotor 40amp

VerimC, 2016-10-12:
Hello I have just purchased Xrotor 40amp esc and the new ZMX fusion x25 motors to go with.  With the latest release of cleanflight they work fine.  However when I flash betaflight they begin to beep all 4 in about 1 second intervals.  Any ideas??

AndersHoglund, 2016-10-12:
Betaflight has ONESHOT125 at 1kHz enabled by default. Your ESCs can probably not handle that. 

On the configuration tab, turn off ONESHOt125 and select PWM. 
Turn on the "Motor PWM speed separate...." switch. 
Select 500hz in the "Motor Speed freq.." box.

mikeller, 2016-10-12:
I think NAZE can handle 1kHz and ONESHOT125 (at least my rev6 can). My guess on what's wrong would be that your ESCs are probably not configured to support ONESHOT125 (which is preferable to using PWM). You should download the manual and configurator for your ESC (from http://www.hobbywingdirect.com/pages/download) and see if you can enable ONESHOT125.

AndersHoglund, 2016-10-13:
The ESC manual says max 621Hz..... And Oneshoe125 should not need enabling, it is automatic. If supported at all.
[imgmgr.banggood.com/images/upload/2014/12/Hobbywing%20xRotor%20Manual.pdf](http://imgmgr.banggood.com/images/upload/2014/12/Hobbywing%20xRotor%20Manual.pdf)

ledvinap, 2016-10-13:
Some ESCs have problem autodetecting OS125 at 1kHz - it may be detected as 1kHz PWM ..

AndersHoglund, 2016-10-13:
Well, this ESC can not do anything above 621Hz. So, std PWM @500Hz would probably work.

martinbudden, 2016-12-20:
Closing, since no further response from original author.

closed 2016-12-20 by martinbudden