betaflight/betaflight #1311: Simple S.PORT / MSP bridge

raphaelcoeffic, 2016-10-13:
As a follow-up on the discussion in #1277, I'd like to propose some sort of simple S.PORT <-> MSP bridge, which should allow for configuring things like PID / LED colors directly from a LUA script in OpenTX.

Please note that the required features in OpenTX have appeared in 2.2. I'll implement it myself if nobody minds.

ledvinap, 2016-10-13:
Implementing SP2UART emulation both on FC and OpenTx would be great. Then just connect MSP to this (virtual) port ..

martinbudden, 2016-10-13:
Yes, this sounds like a good idea.

raphaelcoeffic, 2016-10-13:
@ledvinap I'm afraid this not going to be a full generic bridge, as the SP2UART is, as it is not open, and would probably be quite inefficient, given the small bandwidth available. I'll focus on MSP, so that we get something useful.

ledvinap, 2016-10-13:
It should be trivial to reverse-engineer SP2UART, Saleae trace is probably all that is necessary (both serial and sport side)
FrSky implementation of SP2UART is not perfect, but it is standard, IDs are allocated, it may be used with HW SP2UART on both sized (work may be reused) etc ..

Implementing new protocol is possible, but implementing it cleanly in reusable manner will be hard..

raphaelcoeffic, 2016-10-13:
@ledvinap: well, if you give me the SP2UART A & B ;-) I could possibly borrow one, but not the other... And they are not quite spread in fact. But well, if you have the traces, or the devices, I can give it a shot (I do own a logic analyser).

raphaelcoeffic, 2016-10-13:
@ledvinap: I just a supplier for these, and it would be around 40€ for both (incl. delivery). I don't wether or not it would be worth it, given the fact that they are mainly used as an S.PORT ADC all the time... I've never seen any use-case using the serial transmission, and there is no support in OpenTX, meaning that the LUA script would have to implement that on top of S.PORT on its own. Frankly, I think it's way to roll-my-own.

ledvinap, 2016-10-13:
@hydra: What do you think? Any donation money left?

raphaelcoeffic, 2016-10-13:
@ledvinap: after researching a bit more, it seems to me that this product has been abandoned. The tool which is supposed to be used to change the baudrate (1200bps default) is not even available from FrSky. Let me craft a working prototype, and we can take it from there, if needed.

ledvinap, 2016-10-13:
OK
Personally I would prefer frame-oriented transparent protocol with at least basic error checking/recovery, but it may be overkill for this purpose.

How many IDs to allocate? Guys at OpenTX may help with this.. With 16bit id + 32bit payload, it would make sense to transfer some metadata as part of ID ..

raphaelcoeffic, 2016-10-13:
@ledvinap: well, the range of useable value IDs is quite big. I'm sure I can use quite some of them.

ledvinap, 2016-10-13:
@raphaelcoeffic : Yes, I know. But it is quite rude to other parties involved (FrSky and OpenTX team comes to mind), it may cause problem in future (new sensor will collide) etc ...

Also implementing both sides will require quite an effort - so make it reusable if possible ..

raphaelcoeffic, 2016-10-13:
@ledvinap: no worry, I'll share with my OpenTx buddies. My biggest issue right now is that I cannot seem to get my F4 disco board to fly... I bought today some gyro-breakout board for it (MPU-6050), but I'm still getting HardFaults all the time, without even being able to trigger a breakpoint on main()... But I'll get a naze32 from a friend this weekend.

raphaelcoeffic, 2016-10-14:
@ledvinap: concerning the IDs, I've just found these two in the OpenTX code:
````
#define DIY_FIRST_ID              0x5000
#define DIY_LAST_ID               0x50ff
````
These IDs (only) are passed to LUA's reception queue, meaning that there is nothing we can use anyways. This gives us 1+4 bytes for each frame (low byte from value ID + 4 bytes value). This is 40bits per frame.

ledvinap, 2016-10-14:
@raphaelcoeffic : Great!
Maybe it would make sense to set topmost bit for all messages - lower half of DIY range can be used for other purposes.

raphaelcoeffic, 2016-10-14:
@ledvinap: ok, why not. By the way, the LUA will have to have a sensor ID of its own as well. I understand that BF uses 0x1B, like most custom/non-frsky applications. The other IDs defined in smartport.c do not seem to be used by frsky sensors. However, it would probably make sense to make these sensor IDs configurable somehow (CLI should be perfectly sufficient; just in case the defaults are conflicting with anybody else).

ledvinap, 2016-10-14:
Sensor id is configurable for frsky sensors (servo changer). Using 0x1b should do, cf/by should handle it separately from this PR, if necessary. 

BTW: using multiple ids will increase available bandwidth - polling cycle is all known ids plus one discover packet

raphaelcoeffic, 2016-10-14:
yes, but the FC will have to know which sensor ID the LUA script is going to be using. This means that it either should never ever change, or be configurable. Given the fact that the FC is probably the only smartport sensor on the bus, this is possibly a non-issue. Just tell me what you think is reasonable.

raphaelcoeffic, 2016-10-21:
@ledvinap: do you think it makes sense to switch the normal smart port values off while  using MSP/smart port? That would definitely speed up the process.

ledvinap, 2016-10-21:
Smartport values may be sent with same sensor ID, so it should be enough to decrease update frequency (and even that may not be necessary).

raphaelcoeffic, 2016-10-21:
I know, but still, if I send the MSP packets every 5th or 10th value, that might be quite slow on the other end.

raphaelcoeffic, 2016-10-21:
basically, not cycling through frSkyDataIdTable anymore, as long as my send buffer for MSP data is not empty

ledvinap, 2016-10-21:
Stalling telemetry when MSP packet is ready is good idea.
And maybe insert one telemetry packet after few (~10) MSP ones to prevent total telemetry stalling (some timeout may trigger otherwise), but that's not probably necessary .. 

I have SPort telemetry implemented as priority queue in my CF clone (local branch), using it may help with bandwidth sharing (and IMO is cleaner than current telemetry implementations; but priority head is a bit overkill, sorted list should do)

raphaelcoeffic, 2016-10-21:
Thanks for the hint, I'll have look.

ledvinap, 2016-10-21:
https://github.com/ledvinap/cleanflight/blob/local/src/main/telemetry/sport.c
probably not working now, it diverged too far from upstream ...

raphaelcoeffic, 2016-10-21:
Thx, I'm currently looking at it. I like the send queue, I must say ;-) I want something very similar for MSP.

ledvinap, 2016-10-21:
The send queue is there mainly for multi-packet messages (GPS). Much easier to implement them without complicating code.

jaapp, 2016-10-30:
This idea is amazing. It opens up so many possibilities. Such as leveraging the osd config menu so you have it on the taranis.

Also, for the first time, the flight controller could be aware of the date and time(as set on the taranis). Log files could be time stamped.

raphaelcoeffic, 2016-10-31:
@jaapp: yes, I'd love that (both of them). However, I'm a bit concerned with the achievable bitrate over SPORT. Not sure it would be enough to transmit all the menu names. We might have to trick a bit here.

jflyper, 2016-11-02:
I (temporary?) added a code to traverse the menu contents of the CMS to measure count and total size of the text labels (not counting text as a value).
ATM, there are 103 labels which adds up to 926 bytes (including terminating zero).
Adding intelligent vtx support will likely to add around two dozens of labels and 100 bytes of data.

raphaelcoeffic, 2016-11-02:
> I (temporary?) added a code to traverse the menu contents of the CMS to measure count and total size of the text labels (not counting text as a value).
> ATM, there are 103 labels which adds up to 926 [...]
Well, ok, so that would be around 10 seconds pure download time… We’de better not do that all at once!

DanNixon, 2016-11-02:
What if the user uses BF configurator to save a file to their Taranis SD card which can then be read by the LUA script that provided the client?

This way there'd be local caching without long download times. The file could be versioned and the CMS disabled if the Taranis copy of the strings is out of date.

raphaelcoeffic, 2016-11-02:
I think we should first get both features merged the way they are now, and we can then take it to the next step and try to make them work together, if you don't mind.

jflyper, 2016-11-02:
No prob. That was our original plan :)

jflyper, 2016-11-03:
Enabling S.PORT on telemetry is the only configuration procedure?

If that's the case, the bridge port is not visible from CMS, so there must be a dedicated (pseudo-) displayPort device for CMS to handle the bridge.

raphaelcoeffic, 2016-11-03:
> Enabling S.PORT on telemetry is the only configuration procedure?
> 
> If that's the case, the bridge port is not visible from CMS, so there must be a dedicated (pseudo-) displayPort device for CMS to handle the bridge.
> 
That's right, I'm afraid.

martinbudden, 2016-11-05:
Actually, I see it as an advantage having a displayPort over smartport, and it should be in an independent module. We need to get the CMS code merged before this, but that shouldn't bee too long now.

raphaelcoeffic, 2016-11-05:
@martinbudden I'll start as soon as the LUA config script is done, and the CMS code is merged.

martinbudden, 2016-11-05:
@raphaelcoeffic , great. There is still a bit to do on the CMS code, but it's almost there now.

raphaelcoeffic, 2016-11-05:
No worry, I'm not yet done with the normal setup LUA script either ;-) Maybe a couple of days?

martinbudden, 2016-11-05:
Yes, at least a few days

raphaelcoeffic, 2016-11-08:
[image]
[image]
[image]

llambkin, 2016-11-08:
WOW! This stuff is next level :D GREAT WORK :D

llambkin, 2016-11-08:
Is there any changes to the wiring to make this work @raphaelcoeffic? do we just need the RX telemetry wire connected, or both the TX and RX for s.port?

raphaelcoeffic, 2016-11-08:
> Is there any changes to the wiring to make this work @raphaelcoeffic? do we just need the RX telemetry wire connected, or both the TX and RX for s.port?
> 
If you have Smart Port Telemetry working, you're good to go. OpenTX 2.2 is required as well.

The corresponding LUA script should be ready very soon.

raphaelcoeffic, 2016-11-08:
As long as smart port telemetry is already working for you, you're covered.

raphaelcoeffic, 2016-11-10:
So here is the PoC setup script. It does not cover much functionality, but it works so that PIDs can be configured.

@martinbudden Is it now time to start the integration with the CMS?

martinbudden, 2016-11-10:
@raphaelcoeffic , the CMS changes have all been merged in, so yes, it's now time to start integrating this.

raphaelcoeffic, 2016-11-10:
any hints or pointers? Does smartport need to become a mspPort?

martinbudden, 2016-11-10:
You should create a displayport over your smartport, similar to the way there is a displayport over the `MSP_DISPLAYPORT` command. See https://github.com/betaflight/betaflight/blob/master/src/main/io/displayport_msp.c to get started

If you have any problems, just ask.

closed 2016-12-20 by borisbstyle

olexs, 2017-01-08:
Is there any progress on integrating CMS and Smartport? If I understand the current state of things correctly, the CMS is already usable via MSP (and is being used by MW OSD), and MSP over Smartport works as well; so all that's needed is a Lua script that would access CMS over MSP over Smartport, and show the menu on the Taranis display. I might be willing to dig into Lua to make it work, if no more changes are necessary on the Betaflight code end.

raphaelcoeffic, 2017-01-09:
@olexs some code on the betaflight side is required, as the protocol is not yet prepared for spontaneous messages from the FC. We might need some sort of re-transmissions as well, as CMS over serial does not have any such mechanisms (as far as I know).

olexs, 2017-01-09:
We should probably create a new issue to work on this properly, with "CMS over SmartPort" as a primary focus and issue title. The MSP/SmartPort bridge that was the focus here is largely done, I think.

raphaelcoeffic, 2017-01-09:
@olexs absolutely! After you, mister!

olexs, 2017-01-09:
There we are. I summarized what I think is needed in the issue description.