betaflight/betaflight #1319: rate curve and preview not showing

Rclf, 2016-10-13:
Strange problem on xracer 3.0. The rates preview and the the rate curve are not showing at all, its blank

mikeller, 2016-10-13:
Weird. Can you give me the configurator version (top middle of screen on startup), and check if you get any errors in the console (right-click into configurator, 'Inspect') when you open the PID Tuning tab (the one with the rates curve)?

Rclf, 2016-10-13:
it's 1.8.3 and nope no errors in the console. It must be the fc, my other boards show them normally.

mikeller, 2016-10-13:
Could also be some weird config settings causing the problem. Can you post a CLI `diff all`?

Rclf, 2016-10-13:
ok this is mighty strange, the diff command does not work on this board, i get "unknown command". Just tried it with another board and works fine I get all the info.

blckmn, 2016-10-13:
try version and see if the firmware is correct

Rclf, 2016-10-13:
ok I am reflashing the board to see what happens

Rclf, 2016-10-13:
thanks blckmn I reflahed and seems to be working now.I will test it tommorrow and report back if there is any issue

closed 2016-10-19 by borisbstyle

dubbbya, 2017-01-15:
I have this same issue and I (think) I flashed it properly. SPF3 FC.

mikeller, 2017-01-15:
@dubbbya What version of the firmware are you using?

dubbbya, 2017-01-15:
2017-01-15 @ 15:04:38 -- Flight controller info, identifier: BTFL, version: 3.0.1
2017-01-15 @ 15:04:38 -- Running firmware released on: Oct 18 2016 10:35:52
2017-01-15 @ 15:04:38 -- Board: SRF3, version: 0
2017-01-15 @ 15:04:38 -- Unique device ID received - 0x3100245933570b20383735

mikeller, 2017-01-15:
That's weird, that should work. Can you check to see if you get any error messages in the console (right-click configurator, click 'Inspect', the console is the bottom half of the window that opens) when opening the PID Tuning tab?

dubbbya, 2017-01-15:
Where do I right click? I noticed that when i minimize and maximize my window the animation appears but dissapears when i switch between PID tab and Setup tab I'm in betaflight 1.8.8

dubbbya, 2017-01-15:
The picture appears, and the numbers move, but not the model of the drone

mikeller, 2017-01-15:
Ah. That's a problem with your chrome installation then. Uninstalling chrome and reinstalling it should fix this.

iam2062001, 2017-01-22:
mikeller,
I am getting the same issue
# Betaflight / SPRACINGF3 3.1.0 Jan 16 2017 / 23:39:46 (704b12f)
on setup tab no preview,receiver tab no preview.
the graph on receiver tab is working though.
I reinstalled chrome, no help. It will visually show up if I minimize, but will not follow H/P/R axis Deg.

I dont trust that all my programs on my computer are up to speed, what program is running animation in betaflight, I can try and update or reinstall that driver.

thanks for any help.
David.

mikeller, 2017-01-22:
@iam2062001: As stated above, it is Chrome / chromium.