betaflight/betaflight #1323: Naze32 R6 100% CPU usage

SargeR33, 2016-10-15:
I have a Naze32 r6 in a couple of my quads and only since moving to 3.0 have I seen this issue. If I run telemetry, 1khz 1khz i get 18 - 20%. If I run 2kz 1khz and telemetry I get 80 - 100%, if I switch off telemetry and run 2khz 1khz I get 20%. Disabling accelerometer, I can get 2khz 1khz telemetry at 15%. I am still a noob though and run horizon mode. Is there anything I can do(bar upgrading fc) or should I revert back to 2.9.x? It seems accelerometer is the huge consumer when run with telemetry.

mikeller, 2016-10-15:
The values that you are getting sound reasonable for the not so powerful MCU that is on the NAZE32. Besides reducing your loop frequency, and disabling heavy users like accelerometer, you could also switch to the 'legacy' PID controller in the PID tab, at the cost of getting less stabile flight characteristics. But all in all, upgrading to a controller with a more powerful MCU is the way to get away from having to make compromises.

mice5, 2016-10-18:
i don't know if i'm right, but are you running telemetry on software generated serial? I think only 1 uart was available on F1 boards, so soft-serial telemetry can really slow things down.

SargeR33, 2016-10-19:
Yes I am actually running the telemetry on a soft serial port. I thought this was the case. Running a virtual serial port is resource consumption in it self, telemetry with that would be worse. I have made the move to xracer f303 v3 fcs and sp racing f3's from now on in. I think I will roll this back to a 2.9x release.

blckmn, 2016-10-19:
Closing this as it was related to softserial.

closed 2016-10-19 by blckmn