betaflight/betaflight #1324: MultiShot TARGET=REVO No  motor response with MultiShot active

RobbieHood, 2016-10-15:
Hi BetaFlight,

I have a weird issue today I can not seem to figure out.
Decided to try the MultiShot protocol on my Atmel based Afro 20a's.

Went to the CleanFlight repo to source the BLHeli MultiShot files.

https://github.com/cleanflight/blheli-multishot/releases

Flashed them via BLHeliSuite > flash other > corresponding hex file.
Opened BetaFlight config and enabled the MultiShot protocol then rebooted.
went to the motor tab, hooked up my lipo and got no response.

Verified my files a couple of times and reflashed with same result.
Then I switched the motor protocol back to OneShot125 and the motors responded again.

This is my diff:

# version
# BetaFlight/REVO 3.0.1 Oct  9 2016 / 16:45:21 (505502e)

# name
name HoodX250

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature LED_STRIP

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# led
led 0 4,15:S:AOBI:2
led 1 5,15:S:AOB:2
led 2 6,15:S:CO:2
led 3 7,15:S:CO:2
led 4 8,15:S:CO:2
led 5 9,15:S:CO:2
led 6 10,15:S:AOB:2
led 7 11,15:S:AOBI:2

# color

# mode_color
mode_color 6 0 12
mode_color 6 1 6

# aux
aux 0 0 1 1975 2025
aux 1 27 5 1575 2100
aux 2 1 5 975 1025
aux 3 2 0 1175 1225
aux 4 25 1 1975 2025

# adjrange

# rxrange

# rxfail

# master
set min_throttle = 1020
set max_throttle = 1970
set motor_pwm_protocol = MULTISHOT
set serialrx_provider = IBUS
set deadband = 5
set yaw_deadband = 3
set baro_hardware = AUTO
set blackbox_device = SPIFLASH

# profile
profile 0

set p_pitch = 54
set i_pitch = 40
set d_pitch = 17
set i_roll = 65
set d_roll = 14
rateprofile 0

# rateprofile
rateprofile 0

set rc_expo = 80
set rc_yaw_expo = 45

Not sure if this is something wrong in the BLHeli MultiShot files or in BetaFlight.

borisbstyle, 2016-10-15:
Those escs dont support multishot

RobbieHood, 2016-10-15:
Am I really being this dumb.....

What is it I am flashing here then ?

https://github.com/cleanflight/blheli-multishot/tree/master/Atmel/Hex%20files

borisbstyle, 2016-10-15:
> tree/master/Atmel/Hex%20files

Atmel escs dont support it. Not sure why the file is there

RobbieHood, 2016-10-15:
Oh damn... I am sorry for opening the issue then. I thought I read about them not being supported some months back but now that I am running BetaFlight I just want to test all new codes too run smoother.
Because the files where there I presumed that atmel was supported by now...

closed 2016-10-15 by RobbieHood