betaflight/betaflight #1336: GPS speed on OSD always = 0

airPixel, 2016-10-17:
I need to display GPS speed on OSD... but speed always = 0

setup:
-omnibusF4 with built-in osd
-MTK3333 GPS, 9600, NMEA
-BF 3.0.1_615
-configurator 1.8.3

GPS speed in the configurator shows is true... but on OSD speed always =0

[image]
[image]

example log from GPS:

```
$GPGGA,185925.000,5944.7483,N,03028.6261,E,1,4,4.07,40.2,M,18.0,M,,*68
$GPGSA,A,3,30,02,07,09,,,,,,,,,4.18,4.07,0.96*0C
$GPRMC,185925.000,A,5944.7483,N,03028.6261,E,1.26,41.98,161016,,,A*52

$GPVTG,41.98,T,,M,1.26,N,2.33,K,A*0E
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$GPGGA,185926.000,5944.7484,N,03028.6253,E,1,4,4.07,41.3,M,18.0,M,,*6D
$GPGSA,A,3,30,02,07,09,,,,,,,,,4.18,4.07,0.96*0C
$GPRMC,185926.000,A,5944.7484,N,03028.6253,E,1.23,38.65,161016,,,A*5E

$GPVTG,38.65,T,,M,1.23,N,2.28,K,A*0D
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$PMTKLOG,0,1,a,31,15,0,0,1,0,0*11
$PMTK001,183,3*3A
$PMTK869,2,1*36
$GPGGA,185927.000,5944.7488,N,03028.6247,E,1,4,4.07,42.1,M,18.0,M,,*64
$GPGSA,A,3,30,02,07,09,,,,,,,,,4.18,4.07,0.96*0C
$GPGSV,3,1,11,07,74,179,18,09,52,119,14,30,46,218,37,05,39,298,20*7B
$GPGSV,3,2,11,16,27,043,,23,20,112,,02,13,258,34,27,11,073,*75
$GPGSV,3,3,11,26,06,027,,13,02,280,,21,01,005,*43
$GLGSV,1,1,00*65
$GPRMC,185927.000,A,5944.7488,N,03028.6247,E,1.20,31.21,161016,,,A*5C`
```
```

borisbstyle, 2016-10-19:
We will have a look soon! Thanks for reporting!

borisbstyle, 2016-12-20:
Is this still an issue on the latest builds?

borisbstyle, 2016-12-22:
Closing issue due to lack of feedback. Please reopen issue if it still exists

closed 2016-12-22 by borisbstyle