betaflight/betaflight #1345: Dynamic min_throttle

joshuabardwell, 2016-10-19:
What if min_throttle were dynamically boosted at times when the aerodynamic situation was likely to cause a motor to stall, such as the end of a flip/roll. This would allow you to run a very low min_throttle so you could have long inverted hang time and/or lose altitude quickly, but without risking a death roll at the end of a flip or roll.

The most obvious thing I can think of is to boost min_throttle when the angular rate exceeds a certain value. Alternatively, perhaps it makes more sense to boost min_throttle as a function of rcCommand. Not sure which is better.

rav-rav, 2016-10-19:
I wonder how much difference a higher minThrottle makes for inverted hangtime.
There was a discussion regarding this in the rcgroups thread:
http://www.rcgroups.com/forums/showthread.php?t=2464844&page=1913&highlight=hangtime#post34814377

> Hangtime comes more from power / punch rather from 0 throttle.
> The falling acceleration is anyway higher than a tiny little bit of thrust that props produce.

Slowly rotating props might slow down the quad slightly while it still has upward momentum. Maybe there is a way to measure the impact of different minThrottle?

joshuabardwell, 2016-10-19:
Higher min_throttle definitely makes the copter drop slower when it's right side up.

borisbstyle, 2016-10-19:
I am not sure if that's the right approach. You are just lowering the resolution on this way.
I would rather like to see this solved in the ESC's. Some kind of closed loop should keep track of rpms?

borisbstyle, 2016-10-19:
but this surely easily can be added anyway.......like 2 lines of code to make the min throttle limit dynamic

borisbstyle, 2016-10-20:
Actually the more knowledge I gained about the ESC's te last period this doesnt sound like a bad idea at all.

lunddahl, 2016-10-20:
I would be nice not to have to worry about min throttle, besides configuring it to spin the props reliable on the bench, most people can figure that out. Now for the question of how much to add to be safe from stalls in the air, that is a lot more complicated/difficult, at least it is for me.
With digital protocols on the horizon, if this works, there would not be anything to think about, regarding min throttle and calibrating.
Again With a digital protocol, regarding max throttle and the possible deadband right before that, should that not be the ESC's problem to eliminate that.

borisbstyle, 2016-10-20:
@lunddahl 
Still there is a difference in ESC's on this part. I have some ESC's what never show the issue and some do. You would still have to configure yet another parameter for ESC throttle.

It would be much easier if the ESC had some kind of deadband.

Like for example:
Motors start spinning from 1050, but once reached above 1050 motor would stop spinning again at 1020 or even lower! Once passed 1020 you again need to reach 1050 to make them spin.

On this way you would really not have to worry about it.

Jahnkeanater, 2016-11-02:
Maybe you could tie this into the anti windup code. Whenever the I term is saturated on the pitch or roll axis have the FC hold min throttle up 30 points until the I term calms back down to prevent the oscillation after a flip from causing an ESC desync.

Just something to give the motor a split second to spool up before a demanding maneuver.

borisbstyle, 2016-12-20:
I reviewed this topic recently and spent some time investigating low throttle behaviour on different ESC's. To my surprise this issue only happens on blheli type of ESC's. I tried KISS,  Maverick and IRC ESC's. All of those can easily run with throttle just above the real min_throttle where motors spin. I don't own any other ESC types / firmware to do more comparisation. There is not much more out there anyway

I think the real solution is to have better RPM monitoring in the ESC itself and be able to increase power when RPM's go below critical point (closed loop). Any other way to re-implement this in FC could lead in issues. Automatic adjusting of throttle to this level can even lead to unnecessary oscillations due to auto changing of the throttle in some edge cases.

Therefore I want to close this issue, but feel free to discuss or object.

closed 2016-12-20 by borisbstyle