betaflight/betaflight #1347: PIDsum limit and mixer code

rav-rav, 2016-10-19:
When you take into account the default quad mixer configuration, the PIDsum limit will actually always be (maxThrottle - minThrottle) / 2.
At minThrottle=1050 and maxThrottle=2000 this would be 475. Why is the default value 700?

Lets assume we want to do a fast roll:
`rollPitchYawMix[i] = axisPID[PITCH] * currentMixer[i].pitch + axisPID[ROLL] * currentMixer[i].roll + -mixerConfig->yaw_motor_direction * axisPID[YAW] * currentMixer[i].yaw;
`
```
axisPID[PITCH] = 0 
axisPID[ROLL] = 475
axisPID[YAW] = 0

Motor 0/1: rollPitchYawMix[0/1] = 0 + 475 * -1 + 0; = -475
Motor 2/3: rollPitchYawMix[2/3] = 0 + 475 * 1 + 0;  = +475
```

=> rollPitchYawMixRange = rollPitchYawMixMax - rollPitchYawMixMin; = 950
throttleRange = throttleMax - throttleMin; = 950 == rollPitchYawMixRange

At this point the throttle range is saturated. Motor 0/1 will go to minThrottle and 2/3 will go to maxThrottle. A higher pidSUM will have the exact same result, or do I miss something?

Also the mixer code evolved around min/max throttle integer calculations and will throw out values in that range, which means (maxThrottle - minThrottle) steps.
Considering different output methods to the ESC, especially DSHOT, I think the code should be reworked slightly to allow for 2000 throttle steps.
I'd suggest to switch over to floats and calculate the required mixer output between 0.0 and 1.0 before scaling it to the used protocol.
In order to keep PIDs the way they are, PIDsum could be calculated as (maxThrottle - minThrottle) / PIDsum.

Any thoughts?

borisbstyle, 2016-10-19:
Pidsum limit has nothing to do with total motor output.

Total motor value is a combination of 3 pidsums! Value of 700 gives some more room for other axis.

borisbstyle, 2016-10-19:
See this PR where mixer code will be improved
https://github.com/betaflight/betaflight/pull/1343

rav-rav, 2016-10-19:
Oh, that PR was not there yesterday, I should check everything before posting.

Motor value is the combination of all pidsums of course and is calculated by the quoted piece of code:
`rollPitchYawMix[i] = axisPID[PITCH] * currentMixer[i].pitch + axisPID[ROLL] * currentMixer[i].roll + -mixerConfig->yaw_motor_direction * axisPID[YAW] * currentMixer[i].yaw;`

This is from pid.c:
```
// -----calculate total PID output
            axisPID[axis] = constrain(lrintf(PTerm + ITerm + DTerm), -pidProfile->pidSumLimit, pidProfile->pidSumLimit);
```

So for each axis the limit is pidSumLimit = 700.
If you use more than one axis, the saturation of each motor is even earlier.
Another example for roll/flip at the same time:
```
axisPID[PITCH] = 238 
axisPID[ROLL] = 238
axisPID[YAW] = 0

rollPitchYawMix[0] = 238 * 1 + 238 * -1 + 0; = 0
rollPitchYawMix[1] = 238 * -1 + 238 * -1 + 0; = -476
rollPitchYawMix[2] = 238 * 1 + 238 * 1 + 0; = +476
rollPitchYawMix[3] = 238 * -1 + 238 * 1 + 0; = 0
```
Now motors are saturated even earlier. There's no way to use a pidsum of 700 on any axis without saturating motors.

borisbstyle, 2016-10-19:
Yes saturation always happens easily. In fact the max is 700*3=2100!

But the saturation is being nicely handled as all motors get equally downscaled and the throttle is automatically being set to center so you get the max thrust in both directions.

rav-rav, 2016-10-19:
Can you please explain why it is 700*3=2100?
I'm feeding the mixer in a simulation with different values and looking at the code it's (maxThrottle - minThrottle) at most for one axis or much less if more axis are involved.

borisbstyle, 2016-10-19:
I am just saying that pidsum what goes in motor theoretical max is 2100. Each axis can provide up to 700. Well actually yaw could only do ~500 as yaw has another limiter on P.

As the current mixer translates PID directly into motor values it would mean that one motor could reach 2100 / 2 value, which is 1050. To reach value of 2100 one motor would go -1050 and other +1050.

What you dont want to have is that one axis eats up full motor power so the other axis gets too much destabilisation. That's why I set the limit to a fairly safe level.

What I want to do in the mentioned PR is to remove direct PID to motor relation. Value of 1 will mean full motor power and value of 0 will be min_throttle. On that way the mixer code will be more readable.

borisbstyle, 2016-10-19:
Dont forget that throttle is being added as offset at the end!

Value of 476 in your case means throttle + value

In case of throttle 1600 that would mean 1600 + 476

borisbstyle, 2016-10-19:
So for most people who have min throttle 1050 and max throttle 2000 you only have a range of 950 to mix with.

rav-rav, 2016-10-19:
Actually in the current implementation and mixer config each axis can provide 1400, because the PIDsum of 700 gets multiplied by +1 and -1 for opposing motors making the total range between +700 and -700 => 1400. This is obviously more than the throttle range of 950 and the reason for my examples using 950/2 = 475.

The total throttle range of 950 is what I mentioned in my initial post:
=> rollPitchYawMixRange = rollPitchYawMixMax - rollPitchYawMixMin; = 950
throttleRange = throttleMax - throttleMin; = 950 == rollPitchYawMixRange

Now if your rollPitchYawMixRange is saturated, in the roll example at +475 and -475, this is executed:
```
if (rollPitchYawMixRange > throttleRange) {
        mixReduction = qConstruct(throttleRange, rollPitchYawMixRange);

        for (i = 0; i < motorCount; i++) {
            rollPitchYawMix[i] =  qMultiply(mixReduction,rollPitchYawMix[i]);
        }
        // Get the maximum correction by setting offset to center
        throttleMin = throttleMax = throttleMin + (throttleRange / 2);
}

```
Motors are scaled properly to the throttle range, but because of the saturation, the pilot's throttle can't even be added as offset:
`motor[i] = rollPitchYawMix[i] + constrain(throttle * currentMixer[i].throttle, throttleMin, throttleMax);`

So I think a safe level for the limit should actually be 350 instead of 700.

borisbstyle, 2016-10-19:
Thats why I made it configurable now ;)

Yes but thats being divided by 2 later if you look in the minthrottle and maxthrottle calculation.

Btw try to see real blackbox scenarios and you will never see it above 300 on regular setups

rav-rav, 2016-10-19:
Being able to configure values is always a good. I've not seen many occurrences with PIDsum above 400 in blackbox, which is a good thing as it does not cause instant saturation of motors. That's also a reason however, why the default setting of 700 got me confused.
I have to agree that the whole mixer code will be a lot easier to read once it's in float and between 0.0 and 1.0. Or even -1.0 and 1.0 if negative thrust is possible.

borisbstyle, 2016-10-19:
Going to close this one as the new mixer will solve this all ;)

closed 2016-10-19 by borisbstyle