betaflight/betaflight #1350: New Feature: Full ESC Switch 3D Mode ON/OFF

RipperDrone, 2016-10-19:
To the experts: I was wondering if this is possible (and makes sense to you of course):

Since flashthrough to the ESCs is pretty standard and stable now, could the flight controller soft-swap the ESCs between motor direction "normal" and "bidirectional", e.g. by flipping a RC switch?

Background: Once flying 3D mode, if you switch it off by mode switch, the usable ESC range for acro is then 1500-2000 only, sacrificing on resolution. If the ESCs could be soft-swapped btw normal and bidirectional motor directions, we could have a full blown switch 3D mode on/off implementation.

Any of my reasoning off reality? :-)

ledvinap, 2016-10-19:
Is it really important to have better ESC resolution?

There is feedback loop closed over ESC channel - if there is quantization error due to low resolution, PID will compensate it (switching between two output levels in ideal case; see delta-sigma modulator).

digitalentity, 2016-10-19:
I wonder if we can have as low as 8-16 levels of throttle and something like delta-sigma on ESC side. It's a closed loop system, quantization error will be compensated anyway.

ledvinap, 2016-10-19:
@digitalentity : It may add some delay (or make system more complex from control point) and may increase consumption if ESC tries to brake to lower level. Reducing output level too much will have negative effects.

On the other side, uincreasing resolution over system noise level has IMO almost no effect (and resolution comparable to system noise is probably good target)

IMO the full system is (first order?) delta-sigma - motor power is integrated (to angular rate), this value is compared to reference (in PID) in regular intervals and output value is updated accordingly ..

borisbstyle, 2016-10-19:
I don't know if more resolution will help in any way. I did however notice a significant difference in ESC's of 250 steps of resolution and those with 1000steps, but maybe other things in those ESC's also make it different.

@RipperDrone I am not sure if that is really going to make much difference, but in here more resolution will be gained
https://github.com/betaflight/betaflight/pull/1343

ledvinap, 2016-10-19:
It should be quite easy to discretize values and compare results .. 
```
int x = motor[i];
int res=20;
x += (res / 2);    // round to nearest multiple of res
x -= x % res;
motor[i] =constrain(x, <min>, <max>);  // x may have been rounded out of range
```

RipperDrone, 2016-10-19:
Exciting discussion is spinning up here - thanks guys. I'm probably too much of a noob to have ajudgement on resolution effects, but at least some latest ESC generations brag around with better res than 256 steps so  there might be something in beyond mere marketing buzz...

borisbstyle, 2016-12-20:
@RipperDrone Dshot protocol has increased resolution in 3D. Try that ;)

Closing issues

closed 2016-12-20 by borisbstyle