betaflight/betaflight #1355: Beta Flight Tiny Whoop Issue

bonner1040, 2016-10-20:
Im having some problems with this TinyWhoop. It is demonstrating two unique symptoms that both started at the same time. I flew about 5 batteries after getting it setup this afternoon.

1. The motors just spin as soon as I plug the battery in. Like it is automatically arming. 

2. It wont recognize the transmitter anymore. 

Everything was working fine. I plugged it into Beta flight and changed the min throttle and min command, which I then reset back to normal.

No idea what is wrong. Just flew it around in my apt, bumped it around for sure, but it LOOKS fine.

deepimpactio, 2016-10-20:
It sounds like Brushless/PWM is not selected in ESC/Motor settings. Are you
able to connect to FC with Configurator? I had similar behavior after
upgrading and had to use bootloader procedure to reinstall firmware and
allow me to bind once again.

bonner1040, 2016-10-20:
Thanks. I reflashed it. It noww recognizes the transmitter again.

It does not spin the motor on start up.

I reset the modes in betaflight, and you can see lights on the quad
recognize the arm switch, but it wont arm.

Make it a great day,

Nick Bonner
General Manager
*www.treestuff.com <http://www.treestuff.com>*
cell 440.989.0444
office 877.408.7337

deepimpactio, 2016-10-20:
Have you followed the procedure in the BeeBrain manual? There are also CLI
commands required.

Type in: “set motor_pwm_rate=4000”, hit enter
Type in: “save”, hit enter

If you are seeing your auxiliary inputs on receiver tab and match them to
modes you are programming you shouldn't have any issues.

bonner1040, 2016-10-20:
YAY! I think that worked. I LOVE YOU. I was about to kill myself.

Make it a great day,

Nick Bonner
General Manager
*www.treestuff.com <http://www.treestuff.com>*
cell 440.989.0444
office 877.408.7337

deepimpactio, 2016-10-20:
Good to hear it worked for you brother. Happy flying ;)

martinbudden, 2016-12-20:
Closing, since reported now working.

AndersHoglund, 2016-12-20:
The BEEBRAIN target might benefit from having BRUSHED_ESC_AUTODETECT defined.

closed 2016-12-20 by borisbstyle