betaflight/betaflight #1359: 3D Mode glitches in BF 3.0?

RipperDrone, 2016-10-20:
Have done some further research based upon my former issue report 
https://github.com/betaflight/betaflight/issues/1354

It seems that my ESCs PPM center value is always off 1500, it's around some 1460 instead. I've tried calibrating the ESCs inside BFC (2000us, 1000us, respectively) as well as in BLHeli (1000...2000 as well as 1020...1980 etc.). Regardless what I try, the center value in 3D mode will always end up around 1460-1470. I've set the BFC center value for RC channels to 1500, likewise the 3D Neutral setting.

My ESCs are LittleBees 30A in Bidirectional Mode.

Any ideas? Could it be an offset generated by BF code?

I also noticed that my RS2205 motors spin much faster at full thorttle forward compared to full throttle backward. Maybe this is also due to center offset? Or are those motors wound in a way that they run faster in the supposed sense of rotation than reversed? (at least the RS2205 come in 2 flavors cw and ccw - so far I thought this was only about the threads...)

Your help is appreciated... anybody else having tries 3D and encountered similar probs?

andre92, 2016-10-21:
have you tried adjusting center throttle PPM on the ESCs in BL Heli Suite?
And what happens when you move the slider in the motors tab? There has to be some point that disarms the motors, that will be your center throttle.
You can calibrate the ESCs from 1000 to 2000 as normal.

RipperDrone, 2016-10-21:
Thank you @andre92 for checking into this issue. 

Yes, I have adjusted the center throttle PPM in BLHeli Suite, and it reads back 1500 on point. Nevertheless, motors have the true dead center sitting around 1460-1470.

When I move the sliders in the motor tabs of BLHeli (or BetaFlight), they start out at 1500, but motors are spinning already. I have to reduce throttle to ~1490, then they stop. Reducing further will spin them up below 1450 only. Therefore, dead center is at 1470 for a reason unknown to me.

My ESCs are LittleBee 30A so it's not BLHeli_s code running on them. Anyway, don't see what this has to do with it...

andre92, 2016-10-23:
I did some testing. Although I can not make sure your escs will calibrate to the right value for center throttle, you can still find your esc center value by testing, as you did. You can then change settings in Betaflight configurator to make sure it works correctly. 

On the radio side, the flight controller uses mid_rc as neutral throttle point, with a deadband of 3d_throttle_deadband. By default mid point is 1500 with 50 deadband. So even though esc may see 1460 as mid throttle, it doesn't change your feel. On your transmitter, center is always center

On the ESC side, it uses the values 3d_neutral, 3d_deadband_high and 3d_deadband_low.
3d_neutral is the throttle it sends to the ESCs when disarmed. Set it between 1460 and 1500 if you need to. Then find the values for 3d_deadband_high and 3d_deadband_low.
When flying normal, ESCs receive value above 3d_deadband_high, when flying upside down, throttle stays below 3d_deadband_low. Set these values at an equal value from 3d_center (defaults have 1460 center with 54 space to either side)

RipperDrone, 2016-10-23:
Thank you for doing further testing. I confirm that ESCs have remained stubborn despite all attempts to get them settle at center value of 1500us. However, as stated in your post, BF can be adjusted to manage the ESC neutral setting (which in my case is around 1470) by adjusting 3d_center and 3d deadbands. 

As another option, as per your suggestion, I have artificially increased BLHeli ESC setting to a new value around 1530, thereby enforcing BF to recognize a good fit for 3D_neutral set to 1500. 

Still not sure what is causing this offset but it's a good workaround.

FinalFrag, 2016-11-04:
These are my settings on Aikon 30A BLHeli_S ESCs.

set 3d_deadband_low = 1464
set 3d_deadband_high = 1520
set 3d_neutral = 1492
set 3d_deadband_throttle = 5

In BLHeli I keep them at 1000 - 1500 - 2000

No idea why the neutral is 8 below where it should be, but I can confirm there is something "wrong"

borisbstyle, 2016-12-20:
There was a bug in MSP interface setting wrong neutral or deadband_throttle. Please try 3.1 Alpha builds and new configurator. I have users what reported corrected behaviour.

Open a new issue if it still doesnt work

closed 2016-12-20 by borisbstyle