betaflight/betaflight #1360: Twitches on BF 3.0/3.01 BlueJay F4

mickFPV, 2016-10-20:
Having some twitches on a brand new build.  

Setup: GEPRC VX-5", Xnova Hpers 2205 2300kv, LittleBee 30A-S, Dal 5045x3, XSR SBUS, BlueJay F4 running BetaFlight 3.0.0 with Gyro Refresh 8kHz, PID Loop 2kHz, and MultiShot with BLHeli_S which has been calibrated (through BL Heli suite (also through BF). I'm also running rubber o-rings above and below the board.

I've logged some BlackBox flight data with debug on notch and gyro. Getting peak oscillations at 1400 us throttle (both in flight an on the bench with props off).  Seems to be isolated to the roll and pitch axis as far as the gyro and PID's are concerned.

So far I've tried:

- reflashing ESC's to most current version 16.4

- cycling motor timing med-low, med, med-high

- enabling oneshot125

- I also re-flashed f/c with the latest v. of bf (3.01).

- debugging and enabling a notch filter at peak motor noise frequency

As per other posts:
Solder points are solid, no screws are touching the stators, and not stray wiring to hit the F/C, and in my mind the frame/build is sound. I haven't included a video of the twitches, but it's detectable both through the FPV cam, LOS and an audible motor sound. Any help would be greatly appreciated, short of removing this F/C, I've run out of sensible fixes.

Cheers,
Michael
[blackbox_log_2016-10-20_193048.TXT](https://github.com/betaflight/betaflight/files/541481/blackbox_log_2016-10-20_193048.TXT)

slawek89, 2016-10-21:
You can try to change demag in esc settings to High. I saw simillar problems with motors and this help.

rayone, 2016-10-21:
I have 2 small builds which are very similar and they both twitch **severely**, with motors getting hotter than BF2.9. The only difference in builds are the VTx and one has a PDB and the other doesn't.
Lux F3
RacerStar BLHeli_S (EFM8BB21F16, 48MHZ) on 16.4
RacerStar BR2205 2600KV

I've tried all refresh rates and Multishot/OneShot.
This issue is not present in 2.9, only 3.0 & 3.1.

One quad fell on to concrete from 3m. After this the quad flew normal until the next battery change. I tried dropping it again but I couldn't replicate. This seems like a gyro hw issue affecting yaw, but 2 different quads present it.
Motors are getting hot so I haven't touched "filtering settings" yet.

Logs attached, one was logging at 500hz (couldn't keep up) while just hovering, the other @ 250hz while flying. You can see large blips on yaw.
[LOG00028.TXT](https://github.com/betaflight/betaflight/files/545408/LOG00028.TXT)
[LOG00029.TXT](https://github.com/betaflight/betaflight/files/545409/LOG00029.TXT)

mickFPV, 2016-10-22:
I can also confirm that changing damag value to high was ineffective.  thanks for the suggestion.  I find it interesting that a lot of users see this issue at around low-mid throttle 1300 us.

mickFPV, 2016-10-23:
- tried a 1000 uF capacitor across the terminals with no luck

- finally, pulled the F/C dropped it into a functioning frame which had a Naze32, 20A little bee's and cobras motors, and that also showed severe twitching.

I think at this point this is a hardware issue, not a software related one.  This post can probably be closed off.

closed 2016-10-23 by mikeller