betaflight/betaflight #1371: Telemetry (Realtime)

HorusDrones, 2016-10-22:
Hi betaflight community - this is not so much an issue but an idea I'd like to discuss and develop, if possible. 

I've designed and am building a 'larger' quad for racing and there are some software and features I've found my self suddenly wanting for tuning and perfecting this new beast. 

The quick and short of it is this: due to it's size and the inherent danger involved, tuning, operating and racing will have to be taken very seriously and treated very carefully. I find myself wanting to have a laptop side ground-station with the ability to track the sensors (black-box style) in real-time and have a laptop operator be able to adjust PIDs, filters and so forth in realtime as well. 

Using the UARTS to a typical old 3DR style generic 900mhz or 433mhz radio is both possible and rather easy to do. 

I do know that any changes will not be saved until the bird is grounded, everything is saved and the power is cycled, but this would open a huge range of opportunity for in-race adjustments and performance monitoring/diagnosis. I think this would/could elevate the level of performance and demand placed upon teams particularly in endurance racing and be safer as our rigs get larger. 

I will have my bird up and running within a month or so and I am very serious about this concept. 

All the best
Sherif

mikeller, 2016-10-23:
Hi Sherif. You might be interested in https://github.com/betaflight/betaflight/issues/1311 .

jflyper, 2016-10-23:
I don't think so. He is proposing more versatile and immediate in-flight parameter manipulation. On the telemetry side, it means dozens of temperature sensors on battery, motors, escs, regs, multiple voltage sensors on parallel batteries. On the parameter manipulation side, it means ... well, I don't know, MOSFET gates to turn parallel batteries on and off?

HorusDrones, 2016-10-23:
Even if we start without a dozen sensors, just the blackbox gyro streams with full access to flight control parameters would be ideal. Mikeller, thats a good place to start, but what I'm trying to accomplish is two fold. 1) Remove the dependency on the pilot to actively tune what they are flying. Having a tech that he or she can speak to in order to create a safe tuning and fly experience on larger platforms. 2) Provide full control and monitoring ability to the tech operator in order to help diagnose issues such as prop-wash, oscillations and so forth, without having to land in order to change values, reset and take-off again. 

The next evolution will be to add a range of sensor inputs such as rpm, motor battery and esc temp, voltage, individual motor amp draw etc;.

jflyper, 2016-10-23:
Gyro streams would eat up the bandwidth. How fat is the downlink?

HorusDrones, 2016-10-23:
I think around 250kbps

borisbstyle, 2016-12-20:
Going to close this one.

I believe that realtime telemetry should be done in OSD.

closed 2016-12-20 by borisbstyle