betaflight/betaflight #1375: Connecting to betaflight with SP Racing F3 board

eder0009, 2016-10-23:
Hey All,

I'm having trouble connecting to cleanflight or betaflight with a SP Racing F3 board.  I thought the problem was the FC board so I bought a new one today with the same problem.  I've flown this quad for 6 months with no problems and yesterday put on a furious FPV piggy back OSD and the problems have started.  

The strange thing is I can connect a battery, see video, connect my radio to the quad, see the OSD fine and every thing is fine.  Can't fly it because nothing is set up in betaflight, no way to arm it.  

Yesterday I tried to set up the OSD in betaflight, I wasn't able to figure out what ports to use.  So yesterday I used UART 1 for the OSD and then hooked up the USB and I think that fried the board since on the boards manual, it says not to do both.  Today I'm using UART 2 and flashed a NEW SP Racing F3  board and started to set up the board and it went to crap.  Everything went fine until I tried setting up ports again to set up the OSD.  The board got slow and then stopped responding.  I was able to reflash the board and even connect to it once but now the board is not responding.  I used UART 2 today for smart port (don't know if that is right) and RX on port 3.  I have been connecting to the board via COM 3.

The real problem is that there is not much info on how to set up this OSD.  Only 1 video with no talking.

Any ideas?

jflyper, 2016-10-23:
@eder0009 
It sounds like a support request, which is not handled here.
I will help you in the rcgroup's SP Racing F3 thread.
Can you close the issue once you confirmed the thread?
[SPRacingF3 Racing board](https://www.rcgroups.com/forums/showthread.php?t=2387038&page=170#post36019936)

eder0009, 2016-10-23:
Sounds good,  actually the board to flash using cleanflight last night and got everything set up besides the osd. I'll try setting up the osd with the ftdi and see if that works. Ftdi is coming in the mail

closed 2016-10-23 by eder0009