betaflight/betaflight #1385: USB problems after update

alvarobeiste, 2016-10-23:
Hello, yesterday I updated my brand new Naze 32 Rev6 to Betaflight 3.0, plugged it to the computer and everything worked OK, did this two times as I was setting things up, calibrating ESCs... Everything was good, but today I wanted to set up my receiver and when I plugged my board, the LEDs started working normally, as the FC does every time you connect it, but the PC doesn't recognize it. I have tried everything from shorting boot pads, trying to reflash, uninstalling and reinstalling the drivers... but still nothing, "failed to open serial ports". I also tried with four different USB cables that work in the rest of my controllers, even in another 3.0 Kiss FC, but nothing seems to repair the Naze. Is it something software/firmware related or more probably a human fail when using it?

mikeller, 2016-10-23:
Two questions:
- is it a Rev6 or a Rev6b?
- did you ever plug in a battery while USB was connected?

(The reason for asking this, there is a slim chance on the NAZE32 Rev6 (not Rev6b) that the USB port gets fried by a voltage spike when plugging in the battery while connected on USB. I had this happen to one of mine.)

alvarobeiste, 2016-10-23:
I think it's the normal Rev6, not Rev6b. I think what you are saying is probably right, because when you plug it in now you can feel the USB port and diodes placed near to it getting really hot, as if something was fried there.
I guess you couldn't manage to get it to work again, did you? :)

mikeller, 2016-10-23:
Nope. But I got a free replacement from the supplier, since this seems to be a known (and rare) problem.

alvarobeiste, 2016-10-24:
OK, so I'm going to assume my problem is the same that you had and try to get a replacement... Thanks for your help :)

closed 2016-10-24 by alvarobeiste

martinbudden, 2016-10-24:
@alvarobeiste , if you are replacing a NAZE32, then I would seriously consider getting a more modern board, F3 based at least. Betaflight runs much better on an F3 (or F4) board, and more features are available.

alvarobeiste, 2016-10-24:
@martinbudden Yeah, if I don't manage to get a free replacement I'm going for an F4 board, or maybe another Beta flashed Kiss :)