betaflight/betaflight #1386: OSD performance on REVO target

llambkin, 2016-10-24:
Hello all,

 One thing that the REVO has been suffering from since the first release of 3.0, is that OSD performance is choppy, I have it set to UART6, and when I move the sticks to navigate the OSD, the cursor will jump one or two spaces, It is definitely target related, since I have put different FC's in the same craft, and also tried different Minim's also, I have also tried different firmwares of MWOSD.

This makes it really hard to navigate the OSD, and sometimes resulting in exit, instead of save and exit. I dont know if it is hardware related, or firmware related.

jflyper, 2016-10-24:
@llambkin 
I will take a look at it when #1352 is stable (I wonder if it reproduces on OMNIBUS F4).

BTW, if you are using the OSD config menu on Minim to tweak PID parameters, take a look at the #1352.

jflyper, 2016-10-25:
Yes, it reproduces on OMNIBUS F4 with an external Minim.

It is as if `rcData[]` (BF side) -> `MwRcdata[]` (MWOSD side) update is very slow for some reason.

Let me look into this a bit deeper.

llambkin, 2016-10-25:
Thanks. I did notice that the omnibus OSD works superbly, but didnt
understand why the REVO wasn't the same.

jflyper, 2016-10-25:
Sorry for the confusion.

I tested it with OMNIBUS F4 + MinimOSD, **not** with the integrated OSD, and got a choppy result.

I believe it has something to do with how fast the live RC channel data is transferred from FC to Minim. No details yet; thinking about how to investigate this.

blckmn, 2016-10-26:
It might be the uart - some don't have dma enabled.

llambkin, 2016-10-26:
If that is the case, then that rules out my OSD being the cause for my D-Shot issue.

Although I was having the OSD issue since day one anyway.

-Liam

jflyper, 2016-10-27:
@blckmn MSP messages to OSD are usually small and evenly spaced, so absence of dma would not be a problem. If dma has something to do with this issue, it may be the latency caused by setup time (I don't think it would matter, since it's fraction of a one character time).

jflyper, 2016-10-28:
@llambkin This is just my hunch; what if you enter the MWOSD config menu while connected to the Betaflight configurator?

llambkin, 2016-10-28:
Will give it a go when I get home from work.

llambkin, 2016-10-28:
Ok, tested. No different. It still skips one or two spaces when moving the sticks.

borisbstyle, 2016-10-28:
try without DSHOT

llambkin, 2016-10-28:
Its been this way since release.

llambkin, 2016-10-29:
@jflyper, I would love to try your new code with the injection of the menu. But I am running development branch builds to get D-shot. Is is much effort to make a hex for the REVO for me to test, I will also have to flash my minim to suit right?

jflyper, 2016-10-30:
My 'clogging' problem was (pragmatically) solved (#1422 [comment](https://github.com/betaflight/betaflight/pull/1422#issuecomment-257166195)).

HaseebNain, 2016-11-01:
I have had the UART itself lock up when communicating to the ATMEGA, this issue occurs when working with the OSD (often when switching between values/pages).

The only fix I've had is to reflash the F4 firmware, power cycling the board does not seem to fix the communication.

jflyper, 2016-11-02:
@HaseebNain Can you provide a bit more information?
- Board
- BF rev
- MSP port for OSD
- External OSD hardware and software rev

HaseebNain, 2016-11-02:
Hey @jflyper 

This is a new F4 target that uses the Scarab-OSD codebase. 
Betaflight 1.0.1 for F4 firmware.
We are using UART1 (RX: PA10 and TX: PA9)

An ATMEGA328 is being used for all OSD operation.

This seems to happen whenever the configurator is not connected.

HaseebNain, 2016-11-02:
Planning on using an Arduino to see the information sent from the F4 and see if maybe there is an issue with the way data is parsed by the OSD firmware.

jflyper, 2016-11-02:
Assigning external UART for MSP and use an external Minim is easier. Is that what you mean?

HaseebNain, 2016-11-02:
Yes that's what I meant, sorry about the confusion.

HaseebNain, 2016-11-10:
It seems that the information is not the issue.

I have attempted different speeds, assuming that the F4 is simply too fast for the ATMEGA but saw no improvement over the different preset MSP speeds in config.h

I am wondering if it is worth reworking the MSP in the OSD firmware, perhaps constantly taking in MSP_RC information as it does now, but once the config menu is enabled, only taking MSP_RC data after each action (ie. cursor drawing)

martinbudden, 2016-11-10:
Have you tried changing the serial task priority and period?

Try in `fc_tasks.c` try setting `staticPriority` for `TASK_SERIAL` to `TASK_PRIORITY_MEDIUM` and `desiredPeriod = 1000000 / 200`. I'm not proposing that as the final solution, but changing it will show if it is a priority issue.

HaseebNain, 2016-11-10:
I just tried it and noticed no significant improvement. 

I tried High priority and still had no impact. 

This does not seem to be a priority issue, however it was definitely worth checking out.

martinbudden, 2016-11-11:
I wasn't too hopeful that it was a priority issue, but it's worth eliminating that possibility.

jflyper, 2016-11-11:
I think we should stop the guess work and start working on more technical way.

I would start by hooking up a logic analyzer to (1) PWM port of the S.BUS receiver, (2) UART TX and (3) UART RX. MWOSD will be modified with a code to echo selected channels, or state machine variables on serial. This output will be interleaved with MSP requests, but should have minimum effect.

With this setup, we can examine the timeline of events, receiver pwm change, rcData over MSP over serial and MWOSD's reception. We can compare the timeline between F3 and F4 targets.

I'm away from my equipments until Sunday evening (local time), so if anyone haven't done this, I will do it then.

martinbudden, 2016-11-11:
@jflyper , it would be great if you could do that.

jflyper, 2016-11-13:
Okay. I think there is something wrong with serial output in REVO(-ish) targets.

Target: OMNIBUS F4
Firmware: 3.0.1 Release

Looking at the MSP request-response sequence, responses are periodically corrupted.
Attached is the Saleae capture.
[image]
Here, Channel 02 (REQ) is MWOSD->FC stream, Channel 01 (RESP) is FC->MWOSD stream and Channel 04 (Rcdata rcvd) is a pulse generated by embedding a pair of digitalWrite(HIGH)/digital(LOW) in the MWOSD code after receiving an MSP_RC response.
Corrupt response is the region at +0.50s~+0.55s. Here, all sorts of invalid (and may be valid) responses are stuffed back to back.

I'm suspecting there's something wrong with output buffer management.

[msp_corrupt_f4.logicdata.zip](https://github.com/betaflight/betaflight/files/587979/msp_corrupt_f4.logicdata.zip)

jflyper, 2016-11-13:
The similar corruption can be observed by connecting UART6 to configurator via FTDI and switching to  the receiver tab.

llambkin, 2016-11-13:
Luckily serial RX doesn't talk MSP. Because that would be devastating.

mikeller, 2016-11-13:
From the look of it this is related to #1499.

jflyper, 2016-11-14:
@llambkin I think Serial RX was lucky because it doesn't transmit.

Oh, so, telemetry could be broken, too???

llambkin, 2016-11-14:
Could be. I haven't tested due to lack of inversion on F4. That and I am
using all UARTs. OSD, serial RX and now ESC serial.

mikeller, 2016-11-14:
Now that you're saying it I am getting errors in telemetry anyway, and until now I've attributed that to openLRSng dropping telemetry data if the 19200 baud radio link is saturated, but who knows...

jflyper, 2016-11-18:
I think it's the serial black box that is most violently hit with this one.

mikeller, 2016-11-18:
@jflyper: Probably. But serial BlackBox is more or less legacy anyway, since it cannot keep up with today's loop times.

jflyper, 2016-11-18:
OpenLeger?

jflyper, 2016-11-18:
So, is anyone working on this issue? After d-shot is stable?

llambkin, 2016-11-18:
What? D-shot isnt stable??? Ive even switched back to using my GoPro instead of the Yi.....

HaseebNain, 2016-12-02:
I have had another F4 lock up completely when working with the OSD menu. 

This has only happened when entering the menu of MWOSD but shouldn't cause an error with the F4 and its ability to read serial information. 

Is the corruption leading to the F4 stalling out? Could the buffer overflow cause memory leakage?

jflyper, 2016-12-20:
@HaseebNain I have over looked this one.

Is this problem persisting?

I believe that this problem is nothing to do with MWOSD menu, because the stick is first interpreted by FC then passed to the MWOSD, regardless of the state of the MWOSD menu.

borisbstyle, 2016-12-20:
What is the current state of this issue?

jflyper, 2016-12-20:
I took a look at the tx side queuing code when I found the garbage issue, but couldn't find anything suspicious. Then it was left alone for some reason I can't remember ... may be someone was working with VCP stability issue and didn't want to collide with it.

I'll give another look at the signal in couples of days. Hang on.

HaseebNain, 2016-12-20:
@jflyper It has still has the occasional lockup, I wouldn't think that MWOSD would be causing any sort of freezing. That being said, it definitely seems to occur more when in the menus.

The MWOSD seems to state that "NO DATA" is displayed only when there is no MSP information from the FC for 3(?)ish seconds. 

It could be an issue with the FC reading from the radio receiver. I looked through the serial code and nothing in there seems to point towards the FC not being able to send MSP info.

jflyper, 2016-12-26:
Haven't had a chance to test it yet...

jflyper, 2016-12-29:
The problem is persisting. Tested on OMNIBUSF4, UART6, 4191afd.

[OMNIBUSF4 UART6 Corrupt MSP 2016-12-29 23.30.logicdata.zip](https://github.com/betaflight/betaflight/files/677279/OMNIBUSF4.UART6.Corrupt.MSP.2016-12-29.23.30.logicdata.zip)
[image]

There are three occurrences of corruption, two in 3~4sec and another at 5sec.

jflyper, 2016-12-29:
Allright, it has something to do with TX DMA. This is when TX DMA is disabled.

[OMNIBUSF4 UART6 Clean MSP 2016-12-29 23.30.logicdata.zip](https://github.com/betaflight/betaflight/files/677311/OMNIBUSF4.UART6.Clean.MSP.2016-12-29.23.30.logicdata.zip)
[image]

HaseebNain, 2016-12-29:
That's interesting, I wonder if the selected stream/channel is conflicting with the DMA configuration of something else within Betaflight.

jflyper, 2016-12-29:
@HaseebNain I think it's less likely the DMA conflict. In the test above, only other DMA user is ADC which is on DMA2 ST4. UART6 TX is on DMA2 ST6.

HaseebNain, 2016-12-29:
@jflyper Yea it was a long shot, I was just looking through that now and different streams on the same DMA should not be causing any sort of overlap.

The inconsistency is what's bugging me, there doesn't seem to be a pattern to it. It might be the DMA drivers themselves, maybe something was overlooked when converting them for use on the F4?

llambkin, 2016-12-30:
I don't know if it is related tho, I had an issue with the bluejay F4 the
other day. It would reboot my OSD when I enabled MSP on the serial port. I
was using uart1 and D-Shot, I didn't get a chance to do very thorough
troubleshooting, and changed it out for the Omnibus F4 I had kicking about.

I changed the minim, because I thought it may have been that. It wasn't,
the only thing that would stop it from rebooting was to disable MSP for the
OSD.

digitalentity, 2016-12-30:
F1 targets seem to be affected as well.

closed 2017-07-26 by jflyper