betaflight/betaflight #1396: Defaults for beeper_od and beeper_inverted are wrong for NAZE rev6 in CLI 'diff'

mikeller, 2016-10-25:
This is probably due to `hardwareRevision` being set incorrectly when creating the default values for `diff`.

blckmn, 2016-10-26:
There's a target configuration function that should be executed for defaultConfiguration. Otherwise the default is incorrect.

blckmn, 2016-10-26:
Just need to add a call to that (there's a define USE_TARGET_CONFIGURATION).

mikeller, 2016-10-26:
Ha, that function used to be called in `createDefaultConfig`, but it's not in there any more. Is there a reason that it shouldn't be called in there, @martinbudden?

mikeller, 2016-10-27:
Actually no. When resetting a NAZE today i noticed that the values that it defaults to do not work for the rev6:

```
beeper_inverted = OFF
beeper_od = ON
```

I.e. the defaults are recognised correctly, but when reset to defaults the wrong ones are chosen.

blckmn, 2016-11-19:
https://github.com/betaflight/betaflight/pull/1599 should sort this one

mikeller, 2016-11-19:
This was fixed by @mj666's fix to the init order to put hardware detection before loading / creating the config. Sorry for not closing it.

closed 2016-11-19 by mikeller