betaflight/betaflight #1408: CC3D (CC) bricked after set enable_buzzer_p6 = ON

gamerpaddy, 2016-10-26:
after setting "set enable_buzzer_p6 = ON" the connection froze and i wasnt able to connect anymore over serial / usb.
Had to reflash betaflight using SWD port & STLink adaptor.

tried this on a CC3D (CopterControl) with Betaflight 3.0.

no additional information provided, it just froze and never went back up again. after reflashing it worked again.

AndersHoglund, 2016-10-26:
Weird. Exactly what version was that? 
I tried to reproduce on the 3.0.1 release, no issue. Other than I had to restart BFC after setting buzzer and saving. But that happens, and I don't think it has anything with buzzer to do.

    # get buzzer
    enable_buzzer_p6 = ON
    Allowed values: OFF, ON
    
    # version
    # BetaFlight/CC3D 3.0.1 Oct 18 2016 / 10:03:36 (48b7b4f)

mikeller, 2016-10-26:
Could this be a case of different CC3D clones having slight differences in design?

gamerpaddy, 2016-10-26:
$M>ni$M>lÒÿÿa@
# get buzzer
enable_buzzer_p6 = OFF
Allowed values: OFF, ON

# version
# BetaFlight/CC3D 3.0.0 Sep 11 2016 / 00:28:09 (a21694b)

As soon as i do "set enable_buzzer_p6 = ON" it does not respond anymore and Serial connection will fail on reconnect.

I dont know if its a clone, does the "original" have a s.bus port underneath the PWM input (right next to the DSM port)?  If no, then its a clone.

Im updating to 3.0.1 now, and lets see if it gets stuck again.

blckmn, 2016-11-16:
This is no longer an issue in 3.1 - the P6 is removed in favour of resource mapping.

closed 2016-11-16 by blckmn