betaflight/betaflight #1409: ADC Internal Pull Ups enabled messes with current/voltage sensors (F4 issue)

HaseebNain, 2016-10-26:
I am working on a SPARKY2 variant that uses PA2 as its current sensing ADC GPIO, the F4 has the internal pull up enabled and is messing with the ADC readings.

Looking into system_stm32f4.c in "src/main/drivers" shows (lines 126-128)

    GPIO_InitTypeDef GPIO_InitStructure;
    GPIO_StructInit(&GPIO_InitStructure);
    GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_UP; // default is un-pulled input

I did try removing these pull-ups and got valid ADC readings from PA2 but this brings instability to UARTs and isn't something I want to mess with.

That being said the code in adc_stm32f4xx.c in "src/main/drivers" is written to disable all pull-ups on the ADC pins (lines 131 - 141)

    for (uint8_t i = 0; i < ADC_CHANNEL_COUNT; i++) {
        if (!adcConfig[i].tag)
            continue;

        IOInit(IOGetByTag(adcConfig[i].tag), OWNER_ADC, RESOURCE_ADC_BATTERY + i, 0);
        IOConfigGPIO(IOGetByTag(adcConfig[i].tag), IO_CONFIG(GPIO_Mode_AN, 0, GPIO_OType_OD, GPIO_PuPd_NOPULL));
        adcConfig[i].adcChannel = adcChannelByTag(adcConfig[i].tag);
        adcConfig[i].dmaIndex = configuredAdcChannels++;
        adcConfig[i].sampleTime = ADC_SampleTime_480Cycles;
        adcConfig[i].enabled = true;
		
		
    }

I'm not sure why the pull-ups remain enabled after the ADC is initialized.

@borisbstyle do you know of people having issues with pull-ups/pull-downs on various pins?

HaseebNain, 2016-10-26:
Looks like I was able to manually change the pull-up status by actively setting them to not use any pull-ups.

   	GPIO_InitTypeDef GPIO_InitStructure;
    GPIO_StructInit(&GPIO_InitStructure);
    GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_NOPULL; // default is un-pulled input

    GPIO_InitStructure.GPIO_Pin  = GPIO_Pin_2 | GPIO_Pin_3;
    GPIO_Init(GPIOC, &GPIO_InitStructure);

I'll try and make it more modular to fit the rest of the code base.

closed 2016-10-26 by HaseebNain

mikeller, 2016-10-26:
@HaseebNain: Before you put any work into this, you should look into https://github.com/betaflight/betaflight/pull/1392 and the discussions around it. There is a major effort under way to enable custom resource allocation / configuration in 3.1., and the problem that you are trying to solve will likely be part of this solution.

But your always welcome to contribute, by bringing your use case into the discussion, or by testing, or helping with bringing it all together.