betaflight/betaflight #1413: bounce back after a roll

thatquadguy, 2016-10-27:
Hi, i have an Alien 5inch frame with emax 30a escs, cc3d on 2.9bf and emax 2205 2300kv motors. Im having problems getting it to stop from bouncing back after a hard roll no matter how high i have D set to it still does it, I've ha an F3 board on 3.0bf and still the same problem. Any help to resolve this issue would be appreciated.

Jahnkeanater, 2016-11-02:
Try increasing P instead of D. You may also want to try decreasing I.
You also want to make sure that when centering the sticks you are not over correcting. 
Letting go of the sticks can cause the copter to oscillate because the stick is oscillating.

borisbstyle, 2016-11-04:
Upgrade to 3.0.1 and try default settings.

Usually tuning issue should not be an issue here on github.

closed 2016-11-04 by borisbstyle