betaflight/betaflight #1416: RSSI on Airbot Omnibus F3 Pro missing

RespawnDespair, 2016-10-27:
Hello,

i have purchased the Airbot Omnibus F3 Pro flight controller which is by all accounts an awesome package. What i SHOULD have checked upfront was the RSSI on the OSD.
There is some information, mainly on RCGroups, that should have made me realize this functionality is missing.

I can see in the target.h file for the OMNIBUS on line 144:
```
//#define RSSI_ADC_PIN                PB1
//#define ADC_INSTANCE                ADC3
```

Basically meaning this has been disabled completely.
My question is twofold:

1. WHY is it disabled?
2. Is anybody looking into adding this back in?

If any assistance or testing is required i would be more than happy to help. I'm a programmer by trade, but i have not played with these F3 boards yet, so i didn't want to start hacking blindly.

nathantsoi, 2016-10-28:
there is currently only 1 adc supported and its in use (current sensor and voltage sensor), but if you're not using one of those just define RSSI_ADC_PIN to the pin that your not using, e.g. to use rssi instead of current:

```
diff --git a/src/main/target/OMNIBUS/target.h b/src/main/target/OMNIBUS/target.h
index 7a6967c..0f5794b 100644
--- a/src/main/target/OMNIBUS/target.h
+++ b/src/main/target/OMNIBUS/target.h
@@ -136,7 +136,7 @@
 #define USE_ADC
 //#define BOARD_HAS_VOLTAGE_DIVIDER
 #define VBAT_ADC_PIN                PA0
-#define CURRENT_METER_ADC_PIN       PA1
+#define RSSI_ADC_PIN                PA1
 #define ADC_INSTANCE                ADC1

 //#define RSSI_ADC_PIN                PB1
```

to build this, checkout the master branch and `git reset --hard v3.0.1`, apply the patch and `make OMNIBUS`, it should install the toolchain for you automatically.

then open the latest configurator, load the local file `obj/betaflight_3.0.1_OMNIBUS.hex` and flash

if you're up to the task, enabling more ADCs would be cool

in 3.1 this will be assignable via the `resource` cli command and none of this will be necessary

RespawnDespair, 2016-10-28:
Thank for your reply.

Are you implying that with 3.1 i will be able to use BOTH sensors? Current/voltage and RSSI?
Or just that switching between them will not require code modification??

mikeller, 2016-10-28:
Currently two of them are available, current and voltage. You can repurpose either one of those for RSSI with a one line change (or depending on your RX you could have it send RSSI on a RC channel), or you could put in some more work to enable voltage, current, and RSSI to work concurrently.

RespawnDespair, 2016-10-28:
Yeah, i found the solution of sending RSSI via Telemetry back to the transmitter and sending it back as a channel to the receiver, which then sends it to the FC where we can select the channel as RSSI input. For my current setup this will work, but i will look into modifying the source to allow direct RSSI input from the receiver module itself. Thank you for your comments. This issue can be closed!

mikeller, 2016-10-28:
Yeah that would work. The receivers that I use (openLRSng) allow for the RSSI signal to be sent as one of the PPM / serial channels directly by the receiver, which makes it an attractive option.
If that's not possible, then adding another ADC channel for RSSI is a good alternative, and 3.1 will allow for this.

closed 2016-10-28 by mikeller

jumatic, 2017-01-25:
So can someone build this to use rssi instead of current? Have an F3 Omnibus AIO V2 SD without baro and current. But the omnibus firmware ist set to use current. Cant activate rssi adc via cli etc. If its just a one line change, so please compile one firmware for me. maybe others will need this too. Also the Firmware could be separated like the CC3D. With BUZZER or without. Was hanginga round for days now to complete my quadcopter builds. Also have f4 Omnibus Pro v2. Also current not in use and wanna use the rssi pin on backside as adc. Is it possible?

mickkn, 2017-02-15:
Has this been fixed? I get 41% on my RSSI input from a XSR receiver

mikeller, 2017-02-15:
@mickkn: No, this has not been fixed, since the OMNIBUS F3 does not have a hardware pin that is designated for RSSI input. You would have to build your own firmware and repurpose one of the Vbat / Ibat pins for RSSI, as described above. Where are you connecting your RSSI signal from / to?

nathantsoi, 2017-02-15:
@mikeller, it does have a pin assigned, but it uses a second adc (which, as
far as im aware, is not supported in software, yet), the first adc is
currently set to the current sensor.

@mickkn, you can change the adc assignment and pin for rssi to replace the
current sensor for now or use one of these pre-compiled firmwares:
https://www.rcgroups.com/forums/showthread.php?2792346-Exotic-OMNIBUS

mickkn, 2017-02-15:
Yeah okay, I use Voltage, RSSI and Current sensor. Just the RSSI reading that is wrong. 

Could it work in the OSD if I use the Taranis channel method?

nathantsoi, 2017-02-15:
whats the taranis channel method?

mickkn, 2017-02-15:
As @RespawnDespair wrote. 

You can take the RSSI from Taranis and output it on a AUX channel

nathantsoi, 2017-02-15:
Ahh yeah, cool. Just `set rssi_channel = 16` or whatever channel the rssi
value is injected.

FranzAr, 2017-02-25:
Will it ever be activated? Tried to use RSSI of my L9R RX for OSD and doesn't work. Any workaround?

mikeller, 2017-02-25:
@FranzAr: In terms of resource usage, activating as it is is not a good option right now, since the second ADC will require one of the already scarce DMA channels. At some stage ADC inputs will become mappable, and then it will be possible to remap the ADC input for RSSI to a pin / pad that they aren't using for anything else.

jumatic, 2017-02-26:
Have build an analog rssi for f3 and f4 boards. There is no magic to do this. If someone need firmware i will pm that.

Outlook for Android<https://aka.ms/ghei36> herunterladen

Von: Michael Keller
Gesendet: Sonntag, 26. Februar 00:52
Betreff: Re: [betaflight/betaflight] RSSI on Airbot Omnibus F3 Pro missing (#1416)
An: betaflight/betaflight
Cc: jumatic, Comment

@FranzAr<https://github.com/FranzAr>: In terms of resource usage, activating as it is is not a good option right now, since the second ADC will require one of the already scarce DMA channels. At some stage ADC inputs will become mappable, and then it will be possible to remap the ADC input for RSSI to a pin / pad that they aren't using for anything else.

mikeller, 2017-02-26:
@jumatic: Sounds interesting. Can you elaborate on what you have built? Maybe this could be made into a PR and integrated into Betaflight?

jumatic, 2017-02-26:
Just have made the one line change like nathan said at the top. Done this for F3 and F4 Omnibus. A good thing for all would be, if current sensor and rssi would be defined, but you only can set rssi adc or current sensor adc. If you set rssi adc in gui you cant set current and vice versa. Or separated firmware in the list to flash, like for cc3d. Omnibus Rssi and Omnibus Curr also for the f4 boards- F4SD RSSI AND F4SD CURR.

Outlook for Android<https://aka.ms/ghei36> herunterladen

jumatic, 2017-02-26:
@FranzAr
At the Exotic Omnibus thread you can get firmware with PWM RSSI and other stuff. I have done adc rssi with 3.1.5. Depending on what f3 board you use its just a one line change. With the F3 Pro its more complicated i think. On the F4 pro V2 it works also with a one line change at the small rssi pad on the backside. Dont know which pin it is at the F3 Pro board. But i think must be the same as pwm rssi. So you can define it at target h.

mikeller, 2017-02-26:
@jumatic We are working on making ADC inputs assignable to pins like other resources, but we are not quite there yet. At the moment you have to compile your own firmware to change them. We already have over 100 different boards that we support, so releasing different variants for individual boards is definitely not the way to go.

FranzAr, 2017-02-26:
All sounds great, but I don't want to compile my "my own" firmware, I'll wait until @mikeller and the team have found a solution for this. Keep up up the great work :-)

FranzAr, 2017-03-29:
Anything new on that? Still looking for the ability to use RSSI ADC. Bought a Betaflight F3, but also doesn't work. I'm a bit desperate.

mikeller, 2017-03-30:
@FranzAr: RSSI should work on the Betaflight F3. Maybe check to make sure your Rx actually sends an RSSI signal?

FranzAr, 2017-03-30:
It's a L9R RX and sends RSSI as "constant voltage". Works perfectly on the good old SPRF3 and on a MinimOSD. Can't get it to work on the BF F3, I only get 1% or 100%, nothing in between. Also Signal is inverted. Played around with scaling and inversion, doesn't help.

FranzAr, 2017-03-30:
Found some info her #1715 ff
https://www.rcgroups.com/forums/showthread.php?2795213-Betaflight-F3-Flight-Controller-OSD-PDB-SD-card-BEC-current-sensor/page115#post36894495

mikeller, 2017-03-30:
@FranzAr: #2540 Fixes the inversion, and it should be in 3.1.7.

holybroseph, 2018-01-07:
Was there ever a workaround for ADC RSSI input? I've been trying to make the L9R send RSSI to my Omnibus F3 for a while with no luck. So far, I've enabled "RSSI_ADC" on the configuration tab, soldered the signal wire from the L9R to the fc "RSSI" input, and assigned "resource ADC_RSSI" to A01 (in stead of ADC_CURR on A01). I can see a random RSSI value on the setup tab, but it doesn't change when I turn the tx on or off. I've played around with "RSSI_scale" and inversion with no luck.

jflyper, 2018-01-08:
@holybroseph 
Unfortunately, Omnibus F3's RSSI pad is not capable of ADC. This is a well known bug, and there is no workaround.

holybroseph, 2018-01-11:
Thanks. Is there an easy way to tell if a flight controller is capable of ADC RSSI input before buying it?

jflyper, 2018-01-11:
Forums around the net.

jumatic, 2018-01-11:
I wonder that it shouldnt work. I have got it working. But not with 
resource commands. Changed it in the traget folder and compiled it. instead 
of current adc, I have assigned the pin to rssi adc.

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/email_app_android_sendmail_footer.htm>

--- Original-Nachricht ---
Von: jflyper
Betreff: Re: [betaflight/betaflight] RSSI on Airbot Omnibus F3 Pro missing 
(#1416)
Datum: 11.01.2018, 13:36 Uhr
An: betaflight/betaflight
Cc: jumatic, Mention

Forums around the net.

jumatic, 2018-01-11:
https://github.com/betaflight/betaflight/issues/1416
<https://github.com/betaflight/betaflight/issues/1416>

look here.

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/email_app_android_sendmail_footer.htm>

--- Original-Nachricht ---
Von: jflyper
Betreff: Re: [betaflight/betaflight] RSSI on Airbot Omnibus F3 Pro missing 
(#1416)
Datum: 11.01.2018, 13:36 Uhr
An: betaflight/betaflight
Cc: jumatic, Mention

Forums around the net.

mikeller, 2018-01-11:
@jumatic: That's exactly the thing: The hardware design of this board only allows either voltage / current ADCs or the RSSI ADC to work with Betaflight, but not both concurrently. Most users seem to prefer voltage / current data to RSSI (which can be received in other ways), so that's what's enabled by default.

jumatic, 2018-01-11:
have uploaded actual 3.2.4 for Omnibus F3 with RSSI ADC on Current pin
and also on Rssi pin. Try it
[betaflight_3.2.4_OMNIBUS Rssi Curr Pin.zip](https://github.com/betaflight/betaflight/files/1624032/betaflight_3.2.4_OMNIBUS.Rssi.Curr.Pin.zip)
[betaflight_3.2.4_OMNIBUS Rssi Pin.zip](https://github.com/betaflight/betaflight/files/1624033/betaflight_3.2.4_OMNIBUS.Rssi.Pin.zip)

jumatic, 2018-01-11:
Yeah because most do it with Frsky Rssi smart port i think.
For the others i have uploaded files above.
Also a new Target would be possible. Omnibus Curr and Omnibus Rssi ???
But thats not really handy. I know to many targets ;-) Betaflights supports most boards on the market. No other do that. So the only way is to ask someone if he can compile sth. for you ;-)

jflyper, 2018-01-12:
@jumatic Did you try this?
```
resource add_rssi a1
```

holybroseph, 2018-01-12:
Finally figured it out! I connected the RSSI output from the L9R to the current pin on the board. Then I had to assign the ADC_RSSI pin to A01. Just to be clear, the resource assignments for the ADC pins are as follows: 

resource ADC_BATT 1 A00
resource ADC_RSSI 1 A01
resource ADC_CURR 1 NONE
resource ADC_EXT 1 NONE

Of course, I also enabled the ADC RSSI switch in the "Configuration" tab.
With this, I have RSSI from the L9R and battery voltage!

Thank you both for your support.

jflyper, 2018-01-12:
Happy signals now :)

jumatic, 2018-01-12:
I was totally behind, at previous versions <3.2 adc resources couldnt be 
assigned with resource commands.
So now it is easy via resource commands for everyone to get this working 
and that was actually the problem and is now cleared.
Happy flying everyone.

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/email_app_android_sendmail_footer.htm>

--- Original-Nachricht ---
Von: jflyper
Betreff: Re: [betaflight/betaflight] RSSI on Airbot Omnibus F3 Pro missing 
(#1416)
Datum: 12.01.2018, 1:24 Uhr
An: betaflight/betaflight
Cc: jumatic, Mention

@jumatic <https://github.com/jumatic> Did you try this?

resource add_rssi a1