betaflight/betaflight #1418: beeper after battery connection

vytautas-valaitis, 2016-10-27:
When battery is connected, board (tested on naze rev5, spracingf3) sees voltage rising slowly (about 2 seconds). BAT_LOW and/or BAT_CRIT_LOW beeps once or twice. Maybe some timeout should be introduces before checking voltage.

mikeller, 2016-10-27:
I think what you are hearing are the initialisation beeps. By the time the board has finished initialising the voltage on the Vbat pin should be stable, otherwise your wiring is most likely faulty.

vytautas-valaitis, 2016-10-28:
No, I tested disabling/enabling bat_low and system_init beeper flags. I can see battery voltage rising slowly (2 secs) in betaflight configurator top right. Same thing on Taranis telemetry. It might be filter cap charging, or something.

mikeller, 2016-10-28:
That looks like something is wrong with your wiring. Do you filter on the Vbat connection to the FC?

Telemetry is different, the refresh rates are low for things like battery voltage.

vytautas-valaitis, 2016-10-28:
It's just standard Matek pdb and xm20a escs, no additional filtering.
I used to have audible voltage alert on Taranis just when new battery is connected. It always announced some nonsense. Fixed it by adding 2s delay. Tested with two different fcs (naze, spf3) on 3 or 4 builds. Going to make a video later and post more details on my setup if no one confirms this behavior.

tac07, 2016-11-05:
I think i'm observing the same behaviour here.

Beeping pattern on plugging in the Battery goes something like this:
.......--   .  .  .

Disabling BAT_CRIT_LOW makes the bootup silent.

naze rev5
barebones PDB (no filters or such, only a 5v voltage regulator for the FC)

PR #1479 (2016-11-11): Introduced consistent usage of 'batteryCellCount = 0' to indicate 'no battery present'.
Also fixed problems with battery low / critical warnings coming on when connecting battery.

Also fixed problems with beeper settings for 1S setups.

Fixes #1418, #1476.
closed 2016-11-11 by martinbudden