betaflight/betaflight #1423: gcc loader confused?

jflyper, 2016-10-28:
I'm experiencing strange errors from loader, errors telling that there was an unresolved symbols, but the specified file has nothing to do with the symbol.

See #1422's travis output for example.

Any thoughts???

blckmn, 2016-10-28:
Have you done a clean and then try again?

jflyper, 2016-10-28:
You can't expect cleaner than what travis does, can you?

jflyper, 2016-10-28:
Okay, I could remove the unsolved symbol error; it was actually referenced from display.c.

So, the confused loader problem still remains. What should we do?

blckmn, 2016-10-28:
If you make clean, you ensure you get all warnings.

Have you included all c files needed in mk ? Is a function named wrong compared to header file? Have you got an incorrect function name used somewhere.

jflyper, 2016-10-29:
@blckmn The unresolved symbol was resolved.

Yes, it occurs with make clean; make.

Real problem I want to raise is "Unresolved reference pointed out by the loader is not referencing the symbol at all".

AndersHoglund, 2016-10-29:
OK, what was the problem? Care to share so we all can learn ?

jflyper, 2016-10-29:
@AndersHoglund 
The loader says there is an unresolved symbol X, referenced in file Y, line Z.
The actual unresolved symbol X is referenced in file A, totally unrelated with Y or Z.

ledvinap, 2016-10-29:
@jflyper : This is side effect of  LTO, all source files ale merged into larger blocks, compiled and linked. Reference to wrong .c file is probably a bug, but it's quite understandable ..

Disabling `-flto` (or grepping for the symbol) will help.

And LD stands for linker, not loader ..

AndersHoglund, 2016-10-29:
@jflyper Been trying to recreate the problem, pulled your PR branch and can not find any external references to X. And STM32DISCOVERY builds just fine. Please advice what A means and how to re-create the problem, what commit has the problem and how was it fixed. 

    $ git pull https://github.com/jflyper/cleanflight.git bfdev-osd-cms-separation-poc
    .......
    $ git checkout 1aaf2d9d47808dbd723c0c7f4ca390ea5fbe0bb3
    .........
    $ grep -r displayCmsInit src
    src/main/io/display.c:    cmsDeviceRegister(displayCmsInit);
    src/main/io/display.c:void displayCmsInit(displayPort_t *pPort)
    src/main/io/display.h:void displayCmsInit(displayPort_t *);
    .......
    $ make STM32F3DISCOVERY_clean STM32F3DISCOVERY
    .....
    LD betaflight_STM32F3DISCOVERY.elf
    ./tools/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-size ./obj/main/betaflight_STM32F3DISCOVERY.elf
       text    data     bss     dec     hex filename
      160024    4936   26140  191100   2ea7c ./obj/main/betaflight_STM32F3DISCOVERY.elf
    ./tools/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-objcopy -O binary obj/main/betaflight_STM32F3DISCOVERY.elf obj/betaflight_3.1.0_STM32F3DISCOVERY.bin
    make[2]: Leaving directory '/home/andho/projects/cleanflight/andwho/betaflight'
    make -j ./obj/betaflight_3.1.0_STM32F3DISCOVERY.hex
    make[2]: Entering directory '/home/andho/projects/cleanflight/andwho/betaflight'
    ./tools/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/betaflight_STM32F3DISCOVERY.elf obj/betaflight_3.1.0_STM32F3DISCOVERY.hex
    make[2]: Leaving directory '/home/andho/projects/cleanflight/andwho/betaflight'
    make[1]: Leaving directory '/home/andho/projects/cleanflight/andwho/betaflight'
    Building STM32F3DISCOVERY succeeded.

AndersHoglund, 2016-10-29:
Never mind. Can re-create it at this point in history:

    $ git checkout 8816678333b3e1e2c684e14b0c295893b0204a07

Will look into it.....

AndersHoglund, 2016-10-29:
@ledvinap , You are absolutely correct. Removin lto makes the linker point out the correct file.
Annoying linker bug. Not sure what good -flto does and what harm it would do to remove it....
 
    $ git diff
    .....
    -LTO_FLAGS   = -flto -fuse-linker-plugin $(OPTIMIZE)
    +LTO_FLAGS   = -fuse-linker-plugin $(OPTIMIZE)
    .....
    obj/main/STM32F3DISCOVERY/io/display.o: In function `displaySetNextPageChangeAt':
    /home/andho/projects/cleanflight/andwho/betaflight/./src/main/io/display.c:728: undefined reference to `displayCmsInit'

ledvinap, 2016-10-29:
LTO allows compiling whole program (almost) as if it was one large C file - mailny function inlining over .c file boundaries and other global optimizations. It's fine to disable it for debugging purposes, but production code is considerably better with it ...

jflyper, 2016-10-30:
@AndersHoglund @ledvinap Thanks for looking into this.
It's annoying, but can live with it. May be submit a issue to wherever appropriate?

P.S. I tend to call ld loader. Maybe I'm too old. Lived days with linking loader ;)

AndersHoglund, 2016-10-30:
OK, we keep lto for now. Seems to be related to know issue 10 in the TODO list:
https://gcc.gnu.org/wiki/LinkTimeOptimization

If absolutly needed we could have Makefile debug targets or a debug option without lto.

ledvinap, 2016-10-30:
'DEBUG=1' builds disable LTO ...

AndersHoglund, 2016-10-30:
Hehe, it's already there. And I am blind not to see it.. And you are also blind ;-) The syntax is:

    $ make DEBUG=GDB ....

It actually works too.

    make DEBUG=GDB STM32F3DISCOVERY_clean STM32F3DISCOVERY
    ......
    obj/main/STM32F3DISCOVERY/io/display.o: In function `displayInit':
    ./src/main/io/display.c:718: undefined reference to `displayCmsInit'

jflyper, 2016-11-11:
Okay, I'm closing this.

closed 2016-11-11 by jflyper