betaflight/betaflight #1424: Forcing gcc version considered harmful

stronnag, 2016-10-28:
This change does not encourage testing of latest commits on modern compilers.
````
$ make
make/tools.mk:285: *** **ERROR** your arm-none-eabi-gcc is '6.2.0',
but '5.4.1' is expected. Override with 'GCC_REQUIRED_VERSION' in
make/local.mk or run 'make arm_sdk_install' to install the right
version automatically in the tools folder of this repo.  Stop.
````
OK
````
$ echo GCC_REQUIRED_VERSION=6.2.0 >> make/local.mk
$ make
make/tools.mk:285: *** **ERROR** your arm-none-eabi-gcc is '6.2.0',
but '5.4.1' is expected. Override with 'GCC_REQUIRED_VERSION' in
make/local.mk or run 'make arm_sdk_install' to install the right
version automatically in the tools folder of this repo.  Stop.
````
Oh well ...
````
$ make GCC_REQUIRED_VERSION=$(arm-none-eabi-gcc -dumpversion)
### a bit painful, but works !!!!
````
back to stable ????
````
$ git checkout v3.0.1
error: Your local changes to the following files would be overwritten
by checkout:
make/local.mk
Please commit your changes or stash them before you switch branches.
Aborting
`````
for the record, gcc 6.2.0 is reliable (whereas for iNav at least, earlier gcc 5.x produced dangerously buggy firmware), produces smaller images and has very useful link type checking.

Please reconsider this change, or make it optional, "You're not using the developer's preferred gcc version, try 'make arm_sdk_install' " rather than making building betaflight more difficult for people using current compiler versions.

AndersHoglund, 2016-10-28:
Well, yes could be less strict, maybe.  But you are making it harder than needed to fix
Edit make/tools.mk and change:

    $ git diff make/tools.mk
    ....
    -GCC_REQUIRED_VERSION := 5.4.1
    +GCC_REQUIRED_VERSION := 6.2.0

Reverting back:

    $ rm make/tools.mk
    $ git checkout make/tools.mk

Unfortunatatly launchpad.net has not yet released any prebuilt gcc6 packages, 6.0 is slated for 2016-12-31. https://launchpad.net/gcc-arm-embedded
Where would you suggest we get arm gcc6.2 prebuilt from? For all 3 required platforms, linux, mac and cygwin. I don't think we should compile the tool chain our selfs with "make arm_sdk_install". Do-able, sure...but no.

stronnag, 2016-10-29:
Thank you. I think for my case a `make-bf` alias encapsulating dynamic definition of gcc version is the best approach. 
I'd prefer informing rather than imposing a version, but it's no big deal. 
Closing.

closed 2016-10-29 by stronnag