betaflight/betaflight #1425: Set spektrum_sat_bind

JerryTitus, 2016-10-28:
I'm receiving a "invalid name" whenver I try to use this variable in any form ( null, =0,= 5,=9.

FC= Omnibus F4 (version 0)

Builds= 3.0.0 and 3.0.1

When I do a dump, I don't see it in the listing either....  on other releases it shows up after "set sbus..."

---
set serialrx_provider = SPEK2048
set sbus_inversion = ON
set telemetry_switch = OFF
set telemetry_inversion = ON
---

Regards, thank you for your time.

AndersHoglund, 2016-10-28:
Seems spektrum sat bind is not supported on OMNIBUSF4, not enabled in the target files. Not shure why, but needs to be on a serial port with bi-directional data input line. If connected to serial port 1 that has an sbus inverter sitting in the way, then it is uni-directional and binding is not possible. The AIRBOTF4 is done wrong this way, and from what I have understood OMNIBUSF4 is similar. Have no FC to check and confirm this though, just speculating really.

JerryTitus, 2016-10-29:
It has a spektrum connector, so there was at least intent!

I beeped the spektrum connector signal line out, and it is connected to RX1.

Doggone it.

Thank you for your help!

nathantsoi, 2016-10-29:
With the one spectrum sat I've used, first I bound it to a standard
spectrum rx then just connected it to the FC.

AndersHoglund, 2016-10-29:
OK, same problem as with AIRBOTF4/FLIP32F4 then. 
It would have been better if routed to RX3.

mikeller, 2016-10-29:
You can always cut off the connector and solder the wires onto the UART3 pads. Plus you'll have to build custom firmware for it.

AndersHoglund, 2016-10-29:
Add the same code as for REVO, and rebuild:

    src/main/target/REVO/target.h:#define SPEKTRUM_BIND
    src/main/target/REVO/target.h:#define BIND_PIN                PB11

And use RX3 in some way. Connect directly or cut and patch at the sat connector.

CFBronson, 2016-11-10:
omg really.......

nathantsoi, 2016-11-10:
i think only the ppm/s.bus port has the serial inverter, the uart1 pin on the bottom and the dsm port dont go through the inverter.

https://nathan.vertile.com/blog/2016/10/12/omnibusf4/#receiver

CFBronson, 2016-11-11:
Thanks for the reply but... there is still no way to bind it with the GUI
correct? I have to use a ppm rx to bind the sat first?

borisbstyle, 2016-12-20:
@CFBronson as @nathantsoi already said F4 boards don't have hardware inverters. There is only one RX with hardware inverter for receiver.

closed 2016-12-20 by borisbstyle