betaflight/betaflight #1429: XRacerSPI with BF3.0.1: PPM decoder broken ?

Copter38, 2016-10-29:
On XRacer 3.1 (SPI) target with BF 3.0.1 and PPM receiver all set to default parameters:
- when Multishot protocol is selected in synched mode, the PPM receiver signal traces in BF Configurator show regular spikes of about 300µs amplitude. Jitter of the PPM traces is otherwise normal (~1µs)
[image]

- when Multishot protocol is selected in unsynched 32k, all PPM receiver signal traces show abnormal jitter of about 10µs amplitude, but no spikes anymore
[image]

The CPU load is also behaving reverse when changing from synched to unsyched as changing to unsynched increases CPU load by about 2% when it normally should reduce the load.

The above issue does not exist on BF 3.0.0.

rodgerramjet, 2016-10-31:
I have the same board and noticed I started getting motor chirping (BF 3.0)  on the bench when using multishot synced 8/4. Didn't happen with oneshot125. (BF 3.0) Multishot unsynced 8/4/8 also made the twitches go away... but I still have pretty noisy gyro traces on pitch and roll (but that could be my setup)

@Copter38  would these behaviours tie in with your findings above- they seem to.... apart from you mentioning it didn't happen in BF 3.0 - which it does for me.

Copter38, 2016-10-31:
@**rodgerramjet** I don't think the behaviors you are describing are related to this issue. I rechecked BF3.0.0 against the issue I reported and the PPM readings are all super clean. No spikes at all to be seen and the jitter in the PPM signals is limited to 1µs whether on Multishot synched or unsynched.

Looking at the history of changes, I have noticed that in release #594 (in between BF3.0.0 and BF3.0.1) there were changes made to the X_RACERSPI target that deal with PWM inputs and outputs mapping.
Not sure at all but maybe that's what is causing this strange interaction between the PPM decoder and the ESC outputs. 
By the way, I also can observe the spikes on PPM receiver inputs with BF3.0.1 when on Mulstishot42 or Multishot125. On those 2 protocols however, the spikes seem to appear only in unsynched mode.
I did some further investigation on this and what I can see is, regardless the ESC protocol that is chosen, when in unsynched mode, the higher the Motor PWM speed is, the smaller and more frequent the PPM spikes are...

Copter38, 2016-10-31:
Ok, so in an attempt to track this down a bit further, I have downloaded and tested builds #593 and #594 for the X_RACERSPI target. The changes described in my previous message were introduced in build #594.

The results are pretty conclusive: build #593 is working perfectly (as was BF3.0.0) and build #594 is displaying the reported issue!

Could a developer please look into this. Many thanks in advance!!

ZeeHammar, 2016-11-01:
I am experiencing the same issue on the F303 v3.1 and betaflight 3.0.1 running 8k/8k synced. On the receiver tab in betaflight I get sporadic 20us spikes (+20 and -20) on all channels, including AUX's. The spikes happen regardless of stick position.

I didn't think the issue was with betaflight until I saw this issue. I pulled my D4R out, resoldered everything, re-routed wires, and even tried using an external vreg thinking it was electrical noise or something. 

I will test running unsynced and different looptimes later tonight.

Aileron8, 2016-11-04:
I have been experiencing this issue since I received my first Xracer-V3.1 back on BF2.9, 3.0, and now 3.1. I have always avoided the motor twitches by using MultiShot & cranking LT's up to 8K/2K/32K, unsynced of course.

ZeeHammar, 2016-11-04:
I confirmed the 20us spikes were no longer present on the receiver tab when using OneShot42 synced (8k/8k) and the 3D craft representation is no longer moving around. 

I do seem to now have some motor oscillations at high throttle that I didn't  have before despite having my PID's set quite low. But I suspect that may be unrelated to this issue.

Copter38, 2016-11-09:
The issue originally identified with PPM RX has been tracked down as far as BF3.0.1 goes (thanks to @mmiers). More details on this are available [here](https://github.com/betaflight/betaflight/issues/1227).

However, a quick test of Jenkins alpha builds 667 et 680 for the XRACER_SPI target shows that PPM_RX does not work on these BF3.1 Alphas. These tests were done on firmware with all parameters set to default (i.e. with ESC protocol set to Oneshot125).

borisbstyle, 2016-12-20:
Can you report the current status of this one?

mmiers, 2016-12-20:
The fix for the v3.0.1 tag was to remove PPM3 and PPM4 from the multiPPM
and airPPM arrays.

On the master branch, with the PPM rewrite, the fix is no longer relevant.
On the develop branch the fix is already in place.

I haven't played with the alpha and PPM, so I can't comment on if there are
any issues left with it.

Aileron8, 2016-12-20:
I actually pulled my mangled d4rii thinking my Rx was malfunctioning, and replaced with a direct soldered XSR in a tight build. So further PPM testing would be difficult atm. Sbus is working great though I'm happy to convey.

mmiers, 2016-12-20:
Ok.  I still have my tri with a D4RII.  I'll load it and report.

Copter38, 2016-12-20:
Boris, I am happy to report that alpha build 787 does not display this bug anymore. I have been flying it for a while now on PPM with no visible flaw. On my build, this same alpha also has DShot working perfectly fine! 
As it was working with this alpha, I have not tested any of the more recent builds. Too busy enjoying BF 3.1 and its incredible smoothness !! 
I can do a quick check of latest build if it is of any help but I guess if the issue has been resolved in Alpha 787 then it should still be ok for the actual ones.

Thanks to all for the great work. Looking forward to final release of BF3.1.

closed 2016-12-20 by mikeller