betaflight/betaflight #1430: Betaflight 3.0.1 Colibri Race V2.0 TBS Blhelis 25 ESC (pitch) flip issue

synFPV, 2016-10-30:
Hey Guys,

i have issues on hard flips when i flip fast the copter gets out of control on roll axis is everything fine.
when i do slow flips or loops there is no problem but when i do it fast the copter makes its own moves. 

i run 8khz gyro and 4khz pid with airmode, what could it be?

thanks for help

synFPV, 2016-10-30:
ah and with Multishot

ledvinap, 2016-10-30:
Do you have blackbox log?

borisbstyle, 2016-10-30:
Its the esc what desyncs.

Try upping min_throttle and see if you can make it go away.

Justincrowe, 2016-11-03:
Did upping the min_throttle fix it. i Am having the exact same issue. with multiple builds. min throttle is 1050. spaz out after zero throttle hard flips and rolls. only for a second and i can save it. except for the last time, broke my voodoo vdq210. :(

synFPV, 2016-11-03:
i upped min throttle to 1055 and its all fine now, thanks for the help!

closed 2016-12-20 by borisbstyle