betaflight/betaflight #1431: disarm on impact

S7ud, 2016-10-30:
A side effect of running a finely tuned quad into a relatively solid object is the quad flips out of control and even if you disarm as fast as you can it has already been launched into the air by a PID controller trying to deal with things that are uncontrollable. This has happened to me a couple of times and I'm sure I'm not the only one, 

Can a safety feature be added where the Quad auto disarms, and maybe even apply motor stop, when a G force is so large it can only be an impact from a crash? I know the reaction of the flight controller would be much faster and therefore safer than any human. 

This may just save someone from injury from an out of control quad.

Arakon, 2016-10-30:
This has been discussed a few times before. The biggest issue is the fact that very high rates will also cause huge G-forces, and that quite a few times nowadays, even if you hit a tree or something, the quad may spin around and then catch itself and allow you to save it. Automatic disarm would be a 100% crash in such a case.

This is mainly a case of training yourself and using switch disarm. My automatic reaction upon hitting something or knowing I can't avoid a crash is to flip the disarm switch and lower throttle instantly.

rodgerramjet, 2016-10-31:
How about a mode you can assign to a switch that tells the quad you are landing and to disarm immediately on gyro spike?

then you could have a 2 pos switch with:
* Disarm / Landing mode / (autolevel?)
* Arm / Airmode

S7ud, 2016-11-02:
@Arakon, sure flips and rolls can produce high g-forces but these would be classed an intentional moves and can be ignored. Violent unintentional or uncontrolled g-force spikes could be picked out as a trigger to disarm automatically. 

I know its better not to crash and experienced pilots are much better at this than novices. but hey, if it saves someones face then its worth doing right?

digitalentity, 2016-11-02:
I've been looking into the problem and you really can't distinguish between
intentional and unintentional motion. Beginning/ending any rapid maneuver
from FC point of view is the situation when craft is not doing what pilot
commands it to.

From FC point of view hitting a branch and going into rapid yaw rotation is
indistinguishable from pilot commanding a stop from rapid yaw pirouette -
in both cases quad is yawing fast and pilot is commanding not to.

borisbstyle, 2016-12-20:
Rejecting this feature request and closing issue to prevent cluttering of issues. The feature is not realistic

closed 2016-12-20 by borisbstyle