betaflight/betaflight #1433: Look Ma! There's no "BETAFLIGHT" !!!

jflyper, 2016-10-30:
"Betaflight" and "BETAFLIGHT" are hard coded everywhere.

Is there any symbol or macro that generates this string?

I only found BETAFLIGHT_IDENTIFIER in msp/msp_protocol.h; feel somewhat guilty using this out side the msp.

borisbstyle, 2016-11-03:
Where is it used? There is only one betaflight string I believe. Nothing else

jflyper, 2016-11-03:
There are several.
- [https://github.com/betaflight/betaflight/blob/master/src/main/blackbox/blackbox.c#L1172](https://github.com/betaflight/betaflight/blob/master/src/main/blackbox/blackbox.c#L1172)
- [https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_cli.c#L3611](https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_cli.c#L3611)
- [https://github.com/betaflight/betaflight/blob/master/src/main/vcpf4/usbd_desc.c#L60](https://github.com/betaflight/betaflight/blob/master/src/main/vcpf4/usbd_desc.c#L60)
- [https://github.com/betaflight/betaflight/blob/master/src/main/msp/msp_protocol.h#L68](https://github.com/betaflight/betaflight/blob/master/src/main/msp/msp_protocol.h#L68)

However, number of occurrences is not important. The point I want to make is that there is no consistent way for a "wanna be portable across CF/BF/RF/*F code" to say who I am.

martinbudden, 2016-11-03:
There is now. See PR https://github.com/betaflight/betaflight/pull/1452

closed 2016-11-03 by jflyper