betaflight/betaflight #1437: DSHOT blocks ESC passthrough.

AndersHoglund, 2016-10-31:
Just noticed I had to disable DSHOT and select ONESHOT142 to be able to connect to ESCs and update to blheli_s 16.42. If my REVO is configured for DSHOT150 or 300, I get en error like this:
[image]

borisbstyle, 2016-10-31:
Yes I noticed this too. Lets say that I wasnt surprised :)

mikeller, 2016-10-31:
If dshot causes a problem for the blheli tool, it can easily be fixed in the firmware, by switching to oneshot before entering flasher mode, and switching back when exiting.

AndersHoglund, 2016-10-31:
No surprise, not even strange. Just wanted to make a note about it.

blckmn, 2016-10-31:
yes - we need to update the shutdown esc to include killing off dshot.

llambkin, 2016-11-03:
-or- cant we push firmware over d-shot also? wouldnt it be quicker to digitally send the firmware? Or am I overcomplicating things?

mikeller, 2016-11-03:
@llambkin Not sure if adding a second protocol to flash firmware is useful. Having one protocol that is supported by all hardware seems to be less prone to getting broken. And firmware updates are something that you do not perform very often, so having them take a couple of minutes should be acceptable.

borisbstyle, 2016-11-04:
Actually the firmware update is already digital :D.

llambkin, 2016-11-06:
Is it also blocking escprog? I have just done a build on my REVO, and I cant flash KISS ESCs using the escprog command.

blckmn, 2016-11-11:
This should now be resolved.

AndersHoglund, 2016-11-12:
BLHeli passthrough works fine on REVO now. Thnx.

closed 2016-11-12 by AndersHoglund