betaflight/betaflight #1439: UART6 can not be used as default for SERIAL_RX (F4)

HaseebNain, 2016-10-31:
Trying to create a default config file for a target and cannot set a specific UART as default for serial communication to the radio (Taranis).

I am looking into why this is. 

Having a target specify 

    #define DEFAULT_RX_FEATURE      FEATURE_RX_SERIAL
    #define SERIALRX_PROVIDER       SERIALRX_SBUS
    #define SERIALRX_UART           SERIAL_PORT_USART6

looking into lines 714 -716 in "src/main/config/config.c" nothing seems out of place.

    #ifdef SERIALRX_UART
    if (featureConfigured(FEATURE_RX_SERIAL)) {
        config->serialConfig.portConfigs[SERIALRX_UART].functionMask = FUNCTION_RX_SERIAL;
    }
    #endif

The correct UARTs seem to appear in the configurator

[image]

and UART6 can be configured manually to be used for SERIAL_RX. I used #serial to check the enabled settings and found that the port value is offset (UART6 has an index of '5') I then went into 

void resetSerialConfig(serialConfig_t *serialConfig) in "src/main/config/config.c" and manually placed an enable for serial_rx

    serialConfig->portConfigs[5].identifier = serialPortIdentifiers[5];
    serialConfig->portConfigs[5].msp_baudrateIndex = BAUD_115200;
    serialConfig->portConfigs[5].gps_baudrateIndex = BAUD_57600;
    serialConfig->portConfigs[5].telemetry_baudrateIndex = BAUD_AUTO;
    serialConfig->portConfigs[5].blackbox_baudrateIndex = BAUD_115200;

    serialConfig->portConfigs[0].functionMask = FUNCTION_MSP;
    serialConfig->portConfigs[5].functionMask = FUNCTION_RX_SERIAL;
	
It seems to enable UART when checked through #serial in the cli but the toggle is still not enabled, nor does it seem to be taking input from the controller.

Anything else I seem to be overlooking in terms of enabling this as a default?

mikeller, 2016-10-31:
If you try to set the serial port function in the CLI (`serial 5 64 115200 57600 0 115200`), does it show up in the 'ports' tab? If not it's a configurator issue.

HaseebNain, 2016-10-31:
After setting it in the serial it does not seem to update the configurator, it is also not showing any stick movement in the "Receiver" tab.

mikeller, 2016-10-31:
What do you mean by 'enabled in the configurator'? As in 'configured in the CLI'? All I wanted to make sure that this is indeed a firmware bug, and not a configurator bug.

And yes, setting a default in the firmware should be possible, not sure what is wrong there.

What target is this? If it's a new one, check to make sure `#define USE_UART6` is set and `#define SERIAL_PORT_COUNT` are right and `#define UART6_RX_PIN` are correct.

HaseebNain, 2016-10-31:
Enabled as in turning the toggle switch "on". 

All target defines are correct and as the receiver can be seen communicating with the board when enabled in the "PORTS" tab.

I have modified the functionMask to actively define it for serial comm.

    serialConfig->portConfigs[5].functionMask = FUNCTION_RX_SERIAL;

in the resetSerialConfig() function in "src/main/config/config.c"

when checking #serial in the cli it shows 

    serial 5 0 115200 57600 0 115200

It seems to just be missing the definition somewhere in the reset. I do think this is related entirely to firmware though, as the configurator seems to be working as intended.

mikeller, 2016-10-31:
Stupid question, have you done a full chip erase, or otherwise reset your config to defaults after adding the code to modify the `functionMask`?

HaseebNain, 2016-10-31:
Yep, I've been chip erasing and resetting to defaults just to be sure.

jflyper, 2016-11-01:
AVOID_UARTx_FOR_ related?

HaseebNain, 2016-11-01:
Possibly? I wouldn't think so, only because UART6 can be set to SERIAL_RX through the configurator.

HaseebNain, 2016-11-01:
When actively setting
    serialConfig->portConfigs[5].functionMask = 64; //The value of FUNCTION_RX_SERIAL

#serial in the cli shows serial "2 64 115200 57600 0 115200"

HaseebNain, 2016-11-01:
So I tried setting each UART to FUNCTION_RX_SERIAL in the initialization loop

     uint8_t index;
    memset(serialConfig, 0, sizeof(serialConfig_t));

    for (index = 0; index < SERIAL_PORT_COUNT; index++) {
        serialConfig->portConfigs[index].identifier = serialPortIdentifiers[index];
		serialConfig->portConfigs[index].functionMask = FUNCTION_RX_SERIAL;
        serialConfig->portConfigs[index].msp_baudrateIndex = BAUD_115200;
        serialConfig->portConfigs[index].gps_baudrateIndex = BAUD_57600;
        serialConfig->portConfigs[index].telemetry_baudrateIndex = BAUD_AUTO;
        serialConfig->portConfigs[index].blackbox_baudrateIndex = BAUD_115200;
    }

and then overwrote the function masks using 

     serialConfig->portConfigs[0].functionMask = FUNCTION_MSP;
	serialConfig->portConfigs[2].functionMask = FUNCTION_TELEMETRY_SMARTPORT;

	

    #if defined(USE_VCP)
    // This allows MSP connection via USART & VCP so the board can be reconfigured.
    serialConfig->portConfigs[1].functionMask = FUNCTION_MSP;
    #endif

This allowed for the default to be set, still unsure as to why setting it manually does not seem to work. This is a quick and dirty way to set the UART for SERIAL_RX by default and I'm going to keep looking into it.

mikeller, 2016-11-01:
What is your `SERIAL_PORT_COUNT`, and what `SERIAL_PORT_` defines do you have?

HaseebNain, 2016-11-02:
SERIAL_PORT_COUNT is set to 4.

I did try increasing the number to 6 and saw no effect on the default.

    SERIALRX_UART is set to SERIAL_PORT_USART6

mikeller, 2016-11-02:
Sorry, I copy / pasted the wrong snippet in my last comment. I meant to ask you about the `USE_UARTX` stanzas that you are defining. If you're looking at https://github.com/betaflight/betaflight/blob/master/src/main/io/serial.c#L55-L89 , you can see that these need to be matched to `SERIAL_PORT_COUNT`, or things will not work properly.

Also, it might help if you shared your branch, so others could have a look at what you are doing.

blckmn, 2016-11-11:
https://github.com/betaflight/betaflight/pull/1527

It is because if a serial port is not defined, then there will be a mismatch between the port config index, and the usart identifier enum value.

blckmn, 2016-11-12:
This has been tested as ok on the bluejay. Serial is now the default on it (uart6)

closed 2016-11-12 by blckmn